modelica/ModelicaStandardLibrary #778: Rename "IdealCommutingSwitch" to "IdealCommutatingSwitch"?

modelica-trac-importer, 2017-01-14:
**Reported by leo&#46;gall on 17 Jul 2012 17:17 UTC**
The name of Modelica.Electrical.Analog.Ideal.IdealCommutingSwitch seems to use the wrong word:
"commute" (in German: pendeln/umwandeln)
instead of 
"commutate" (in German: kommutieren/umschalten).

Any native speakers here who could comment on this?

Renaming would have to wait for a conversion script, therefore I set milestone to MSL 4.0.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/778

modelica-trac-importer, 2017-01-14:
**Changelog removed by otter on 18 Jul 2013 12:09 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2017-01-29:
@ceraolo What is IEC's wording here?

ceraolo, 2017-01-30:
there is not much on this in the IEV.

After some research there I would conclude that better than "Commuting switch" it is "SelectorSwitch: 
http://dom2.iec.ch/iev/iev.nsf/display?openform&ievref=151-12-24

 And better than "IntermediateSwitch" is "ReversingSwitch":
http://dom2.iec.ch/iev/iev.nsf/display?openform&ievref=151-12-25

beutlich, 2019-07-15:
@AHaumer @christiankral @dietmarw Should we do the renming as propposed by @ceraolo?

- IdealCommutingSwitch -> IdealSelectorSwitch
- ControlledIdealCommutingSwitch -> ControlledIdealSelectorSwitch
- IdealIntermediateSwitch -> IdealReversingSwitch
- ControlledIdealIntermediateSwitch -> ControlledIdealReversingSwitch

AHaumer, 2019-07-15:
To be honest: "Commuting" is wrong.
to commute = German "pendeln", ie. travel from home to work and back.
commutate = German "wenden, umpolen" - this is used in a DC machine,
I'd propose: Commuting -> TwoWay or ChangeOver (German "Wechselschalter")
Intermediate is correct (German "Kreuzschalter"), 
The alternative TwoWayDoublePoleRevesing is clumsy.

beutlich, 2019-07-15:
> Any native speakers here who could comment on this?

@harmanpa Can you please help out on this language issue? Thanks.

christiankral, 2019-07-15:
* SelectorSwitch seems to cover the type of switch we are taking about, but to me the affected model is a special selector switch whith just two switching positions. Selector switch sounds to me like having more than two switching positions. I am in favor of this proposal if some other native speakers agree to it. 
* ReversalSwitch seems the right term to me.

AHaumer, 2019-07-15:
I have the same opinion as @christiankral regarding "Selector", but I don't understand "Reversal" = "umkehren"? I did look up in "Langenscheidt", finding the terms mentiones by my previous post.
I would keep "Intermediate" = "Kreuzschalter", but change "Commuting" to "TwoWay" = "Wechselschalter",

harmanpa, 2019-10-06:
I'd favour the suggestion from @ahaumer of TwoWaySwitch, it is clear and easier to spell than Commutating!

beutlich, 2019-10-06:
So it will be:

- IdealCommutingSwitch -> IdealTwoWaySwitch
- ControlledIdealCommutingSwitch -> ControlledTwoWaySwitch
- IdealIntermediateSwitch: keep as is
- ControlledIdealIntermediateSwitch: keep as is

PR #3135 (2019-10-09): Fixes #778 by renaming "Commuting" to "TwoWay" 
Also includes conversion script and test.
Note: The ticket suggests to replace "IdealCommuting" with  "TwoWay" instead of "IdealTwoWay" which is probably an oversight and would introduce inconsistency.

Fix #778.
commit 4171ae2a (2019-10-09): Merge pull request #3135 from dietmarw/t778-renameSwitch

Fix #778 by renaming "Commuting" to "TwoWay"
commit 6348e322 (2019-10-09): Fixes #778 by renaming "Commuting" to "TwoWay" 

Also includes conversion script and test.
Note: The ticket suggests to replace "IdealCommuting" with  "TwoWay" instead of "IdealTwoWay" which is probably an oversight and would introduce inconsistency.
closed 2019-10-09 by beutlich