modelica/ModelicaStandardLibrary #813: Remove classes that were obsoleted because of name fixes and provide coversion scripts

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 11 Sep 2012 16:42 UTC**
This is a collection ticket for all those classes that need to be removed because they were simply introduced for backward compatibility and which normally extend from `Modelica.Icons.ObsoleteModel`. Typical examples are classes that were misspelled or simply renamed.

Models to be removed and converted so far:
```diff
- `Modelica.Mechanics.MultiBody.Sensors.TansformAbsoluteVector`
+ `Modelica.Mechanics.MultiBody.Sensors.TransformAbsoluteVector`

- `Modelica.Mechanics.MultiBody.Sensors.TansformRelativeVector`
+ `Modelica.Mechanics.MultiBody.Sensors.TransformRelativeVector`
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/813

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2019-06-26:
See the [wiki](https://github.com/modelica/ModelicaStandardLibrary/wiki/Changes-targeting-MSL-v4.0.0-requiring-conversion) on how to remove/convert obsolete classes for MSL v4.0.0.

closed 2019-07-08 by beutlich