modelica/ModelicaStandardLibrary #814: Duplicate definitions of basic electric elements present in Spice3

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 11 Sep 2012 16:54 UTC**
I noticed that at least these basic elements are basically duplicates of the ones provided by
`Modelica.Analog.Basic` library:

 * `Modelica.Electrical.Spice3.Basic.Ground`
   * -> `Modelica.Analog.Basic.Ground` 
 * `Modelica.Electrical.Spice3.Basic.R_Resistor`
   * -> `Modelica.Analog.Basic.Resistor` 
 * `Modelica.Electrical.Spice3.Basic.L_Inductor`
   * -> `Modelica.Analog.Basic.Inductor` 
 * `Modelica.Electrical.Spice3.Basic.C_Capacitor`
   * -> `Modelica.Analog.Basic.Capacitor` 

Maybe I'm not seeing the difference but if not those classes should be removed and references updated in the MSL and a conversion script given for used libraries.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/814

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

dietmarw, 2019-02-19:
@JoeRiel Could you take a look and tell if you agree?

beutlich, 2019-04-08:
@JoeRiel Friendly reminder.

beutlich, 2019-07-13:
@christophclauss What is your proposal for MSL v4.0.0?

beutlich, 2019-10-28:
Removing milestone after missing feedback.