modelica/ModelicaStandardLibrary #825: Default parameters in Modelica.Electrical.Spice3

modelica-trac-importer, 2017-01-14:
**Reported by alessandro&#x2e;picarelli on 18 Sep 2012 09:50 UTC**
 Modelica.Electrical.Spice3.Semiconductors.M_NMOS default model card parameters such as PHI have a default parameter of -1e40V whereas the comment says the default is 0.6V. This is a massive difference in value as per VTO, KP, GAMMA, RD, RS, CBD, CBS, TOX, NSUB, TNOM. Could the default values be brought to reflect those stated in the relative comments to ensure a more robust default parameterisation of these and other Spice3 components. 

A list of equivalent parameters to those in LTSpice would also be beneficial.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/825

modelica-trac-importer, 2017-01-14:
**Comment by majetta on 24 Sep 2012 11:27 UTC**
The models of Modelica.Electrical.Spice3 were extracted from the original SPICE3 simulation tool. In this SPICE3 simulator the calculation of voltage and current depends on whether a parameter was set by the user or is a default value. Since in Modelica we do not have the possibility to check if a parameter was set or if it is the default value, we introduced the -1e40 values and check during the simulation, if the parameter still has this -1e40 value. If so, we assume that the parameter was not set by the user and we give it its real value which is 0.6V in the case of PHI. The concept is explained more into detail here: https://www.modelica.org/events/modelica2009/Proceedings/memorystick/pages/papers/0019/0019.pdf

modelica-trac-importer, 2017-01-14:
**Comment by alessandro&#x2e;picarelli on 24 Sep 2012 16:42 UTC**
Hi, unfortunatelyin the above attachment if I leave the value of NSUB in the mosfet unaltered (-1e40) in the results I then see the value is -1e300. Is this the actual NSUB value also used in the equations?
Many thanks,
Alessandro

modelica-trac-importer, 2017-01-14:
**Comment by clauss on 22 Jan 2013 11:01 UTC**
When analysing your example I found that internally in the equations the correct default value (zero for NSUB) is used. Since NSUB of the modelcard is a parameter which is never changed, -1e40 should be displayed if NSUB is not altered. I suppose the value -1e300 you saw is an display issue from Dymola. I write to the Dymola support.

modelica-trac-importer, 2017-01-14:
**Comment by clauss on 24 Jan 2013 11:11 UTC**
Dymola support answered: It is a Dymola display issue indeed.

You must store the results using double precision in order to represent such
large numbers. Do that by "Simulation > Setup... > Output > Format" and check
the box for "Double precision".

Then you see the unchanged value of NSUB, and -1e40 is displayed.

When you leave the NSUB parameter unaltered, the model regards NSUM as "not given".
If a parameter is "not given" then the model does separate calculations as simply using
the default value. For that reason the (not beautiful) way with using -1e40 was chosen.
We will remove that way if a better solution of recognizing "is given" is available.
Later on I will add an example to this ticket.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2019-07-13:
> We will remove that way if a better solution of recognizing "is given" is available.

I believe this still is not possible by Modelica Specification 3.4. Therefore I propose to move the milestone. Do you agree @christophclauss ?

> Later on I will add an example to this ticket.

This seems to be missed out.