modelica/ModelicaStandardLibrary #883: Make Modelica.ComplexMath naming consistent

modelica-trac-importer, 2017-01-14:
**Reported by pharman on 21 Oct 2012 08:09 UTC**
The functions in Modelica.ComplexMath have a mixture of naming conventions, some match the built-in operators that they redefine for Complex, such as `sin(x)`, some add 's around the name, such as `'abs'(x)`.

According to #880 this is due to lookup from the enclosed Vectors package, however since direct-access to top-level packages with . was added there is no reason that built-in operators cannot be directly accessed such as `.abs(x)`.

It would therefore be beneficial to make the naming in ComplexMath more consistent and remove the 's.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/883

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

balodja, 2019-12-01:
Nice  suggestion. Any progress on that? :)

beutlich, 2019-12-01:
It's addressed in #3047. So you might want to follow that pull request.

closed 2020-01-16 by beutlich