modelica/ModelicaStandardLibrary #966: Fluid names for density and viscosity

modelica-trac-importer, 2017-01-14:
**Reported by rfranke on 17 Jan 2013 19:39 UTC**
Modelica.Fluid generally uses the names `rho` for density and `mu` for dynamic viscosity. See e.g.:
```
Modelica.Fluid.Machines.Pump
Modelica.Fluid.Pipes.BaseClasses.FlowModels
Modelica.Fluid.Fittings.BaseClasses.massFlowRate_dp_and_Re
```
The models 
```
Modelica.Fluid.Interfaces.PartialPressureLoss
Modelica.Fluid.Fittings.SimpleGenericOrifice
```
internally use the names `d` and `eta` instead. These alternative names are not used in Fluid because:
- `eta` is reserved for efficiency (see e.g. Pump)
- `d` is ultimately short, complicating maintenance, and might be messed up with diameter
The uses if `d` and `eta` for density and dynamic viscosity shall be replaced with `rho` and `mu`, respectively. 

Even though these names are only used internally, a renaming in MSL3.2.1 might break compatibility, e.g. for models extending from `Modelica.Fluid.Interfaces.PartialPressureLoss`.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/966

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2019-07-19:
@rfranke Modelica.Fluid.Interfaces.PartialPressureLoss uses the names `d_a`, `d_b`, `eta_a` and `eta_b`. Same idents are also used in Modelica.Fluid.Fittings.BaseClasses.Bends.CurvedBend.{massFlowRate, pressureLoss},  Modelica.Fluid.Fittings.BaseClasses.Bends.EdgedBend.{massFlowRate, pressureLoss} and Modelica.Fluid.Fittings.BaseClasses.Orifices.ThickEdgedOrifice.{massFlowRate, pressureLoss} as function arguments. Shall they be renamed as well?

Furthermore, `eta` is used in Fluid.Dissipation for dynamic viscosity. Shall it be replaced as well?

beutlich, 2019-10-05:
No reply from libray officers -> removing milestone.