modelica/ModelicaStandardLibrary #967: Media shall define type PressureDifference

modelica-trac-importer, 2017-01-14:
**Reported by rfranke on 17 Jan 2013 20:20 UTC**
Fluid generally uses types defined in Media for the medium at hand, like:
```
Medium.Temperature
Medium.SpecificEnthalpy
Medium.AbsolutePressure
Medium.MassFlowRate
```
This enables the approprate definition of attributes like min, max, nominal, default together with a selected medium model. Moreover the medium decides on preferred states, like temperature vs. specific enthalpy.

There is one basic type missing in the medium model:
```
Medium.PressureDifference
```
Fluid 3.2(.1) inconsistently uses `Medium.AbsolutePressure`, `SI.AbsolutePressure` or `SI.Pressure` instead. A consistent use of `SI.PressureDifference` would be feasible, but break with the advantages of using specific medium types, e.g. for the provision of appropriate nominal values.

A medium model already defining preferred states and `Medium.MassFlowRate`, should also define `Medium.PressureDifference`.

One could even think of going one step further: why not defining a new medium model with appropriate type attributes for an application at hand, instead of going through the Advanced tabs of all component models?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/967

modelica-trac-importer, 2017-01-14:
**Comment by hubertus on 6 Mar 2013 15:09 UTC**
changed to milestone 3.3: it cannot be used in a backwards compatible way, delaying to good point in time.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2019-07-01:
@hubertus65 Can you please elaborate what exactly is to be done and how expensive it is to do after all? Thank you!

beutlich, 2019-10-03:
Once again, @hubertus65 Can you please elaborate what exactly is to be done and how expensive it is to do after all? Thank you!

beutlich, 2019-10-28:
Removing milestone after missing feedback.