modelica/ModelicaStandardLibrary #978: Polynomial functions

modelica-trac-importer, 2017-01-14:
**Reported by kdavies on 31 Jan 2013 01:53 UTC**
This ticket is to add a Polynomials package to Modelica.Math.  Unfortunately, after writing the package for another library, I realized that Modelica.Media.Incompressible.TableBased.Polynomials_Temp seems to have the same functions and probably more.  Maybe the two packages can be combined.

One advantage of the new Polynomial package is that it generates polynomial expressions in a nested form (((... a2)\*x + a1)\*x) which should be more computationally efficient.  Also, the tests (added to ModelicaTest.Math) may be more thorough; they are based on the der() operator as recommended in the Dymola 7.4 User's Manual Volume 2.

Regardless, it may be good to move Polynomials_Temp to the Math package so that users can find it more easily.
z

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/978

modelica-trac-importer, 2017-01-14:
**Modified by beutlich on 12 Apr 2015 20:28 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by otter on 12 Dec 2015 06:20 UTC**
Thanks for your proposal. I do not fully understand the implementation (and there is not much documentation).

Note, besides Modelica.Media.Incompressible.TableBased.Polynomials_Temp there is also a more powerful implementation: Modelica_LinearSystems2.Math.Polynomial. The latter uses operator overloading to simplify working with polynomials and provides many operations (including "roots"), e.g.
```
  import Modelica_LinearSystems2.Math.Polynomial;
  Polynomial t=Polynomial.x();
  Polynomial p = t^2 + 8*t + 16;
  Polynomial der_p = Polynomial.derivative(p);
  Real y = Polynomial.evaluate(der_p, 3);
```

This Polynomial definition is used as basis to formulate transfer functions and have operations on transfer functions.

My goal was to include this Polynomial definition in MSL. However, this requires that all relevant Modelica tools support operator overloading (which seems currently not the case).

In order to no longer wait, one could follow your route and combine it with the Polynomial_Temp definitions and introduce a Polynomial without operator overloading (and at some time, another definition with operator overloading). If we follow this, your propposal seems to be not yet sufficient (and I do not understand yet all details), and documentation and test modells are missing. So, at the current stage, it cannot be included in MSL.

For these reasons, I have moved the milesteon.

As a side remark: Can you explain the use case, to specially support polynomials where the lower order coefficients are all zero and have special numerics to compute this in a more reliable form. I am not sure whether your implementation is really more reliable as the Horner scheme used in Polynomial_temp. Are there any publications?

tbeu, 2017-01-18:
Point to @kdavies4.

beutlich, 2017-08-22:
> My goal was to include this Polynomial definition in MSL. However, this requires that all relevant Modelica tools support operator overloading (which seems currently not the case).

The above example code will be supported by the next release of SimulationX.
```mo
model T978_1
  import Modelica_LinearSystems2.Math.Polynomial;
  Polynomial t=Polynomial.x();
  Polynomial p = t^2 + 8*t + 16;
  Polynomial der_p = Polynomial.derivative(p);
  Real y = Polynomial.evaluate(der_p, 3);
  annotation(uses(Modelica_LinearSystems2(version="2.3.4")));
end T978_1;
```
@FrenkelJ FYI

PR #2871 (2019-04-10): Move Polynomials from Media to Math
When Polynomials were added to Modelica.Media.Incompressible.TableBased they were considered temporary. This is a cleanup to move the existing Polynomials package to Modelica.Math (as proposed by #978).

In #978, there has been some intent to introduce the Modelica_LinearSystems2.Math.Polynomial package instead, but given the current limited resources and requirements, it is enough to do this cleanup.

Close #978.
closed 2019-04-10 by beutlich