modelica/ModelicaStandardLibrary #1539: Unused models in ModelicaTest library

modelica-trac-importer, 2017-01-15:
**Reported by sjoelund&#x2e;se on 14 Jul 2014 14:48 UTC**
The attached list of classes are unused by any experiments in ModelicaTest and Modelica. Many of them are due to testAllFunctions not being an experiment, and the testing of functions not being part of models; but there are some more models in there.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1539

modelica-trac-importer, 2017-01-15:
**Comment by otter on 14 Jul 2014 15:59 UTC**
Replying to [ticket:1539 sjoelund.se]:
> The attached list of classes are unused by any experiments in ModelicaTest and Modelica. Many of them are due to testAllFunctions not being an experiment, and the testing of functions not being part of models; but there are some more models in there.

Partially fixed in eaa6a18f4ad45992a677af355846ae9624c98a10 by adding a model ModelicaTest.TestAllFunctions with an experiment annotation that calls most function tests.

Please, produce an updated list of unused models/functions.

modelica-trac-importer, 2017-01-15:
**Comment by sjoelund&#x2e;se on 15 Jul 2014 09:39 UTC**
I re-ran this for trunk, and I will highlight the unused classes in Examples packages (although I guess the UserDefinedGravityField is because OpenModelica does not support this language feature yet):
```
Modelica.Electrical.Analog.Examples.Utilities.RealSwitch
Modelica.Electrical.Digital.Examples.Utilities.RSFF
Modelica.Electrical.Digital.Examples.Utilities.DFF
Modelica.Electrical.Digital.Examples.Utilities.Adder
Modelica.Electrical.Spice3.Examples.Spice3BenchmarkMosfetCharacterization.SpiceConstants
Modelica.Magnetic.FluxTubes.Examples.Utilities.CoilDesign
Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_DOL
Modelica.Mechanics.MultiBody.Examples.Elementary.UserDefinedGravityField
Modelica.Mechanics.MultiBody.Examples.Elementary.Utilities.theoreticalNormalGravityWGS84
Modelica.Mechanics.Rotational.Examples.GenerationOfFMUs
Modelica.Mechanics.Rotational.Examples.Utilities
Modelica.Mechanics.Rotational.Examples.Utilities.DirectInertia
Modelica.Mechanics.Rotational.Examples.Utilities.InverseInertia
Modelica.Mechanics.Rotational.Examples.Utilities.SpringDamper
Modelica.Mechanics.Rotational.Examples.Utilities.Spring
Modelica.Mechanics.Rotational.Examples.Utilities.SpringDamperNoRelativeStates
Modelica.Fluid.Examples.AST_BatchPlant.BaseClasses.setReal
Modelica.Fluid.Examples.AST_BatchPlant.BaseClasses.ControllerUtilities.Adapter_Inference
Modelica.Fluid.Examples.AST_BatchPlant.BaseClasses.ControllerUtilities.Adapter_Superposition
Modelica.Fluid.Examples.AST_BatchPlant.BaseClasses.ControllerUtilities.Block_Recipe_TBD
Modelica.Fluid.Examples.AST_BatchPlant.BaseClasses.ControllerUtilities.BlockMain
Modelica.Fluid.Examples.AST_BatchPlant.BaseClasses.ControllerUtilities.Buffer_Recipe_TBD
Modelica.Fluid.Examples.AST_BatchPlant.BaseClasses.ControllerUtilities.BufferMain
Modelica.Fluid.Examples.AST_BatchPlant.BaseClasses.ControllerUtilities.Port_IdleTanks
Modelica.Math.Matrices.Examples
Modelica.Math.Matrices.Examples.solveLinearEquations
Modelica.Utilities.Examples.calculator
```

modelica-trac-importer, 2017-01-15:
**Comment by sjoelund&#x2e;se on 15 Jul 2014 09:45 UTC**
Also, I should probably say how the list was generated: It is basically the same functionality as `saveTotalModel` in OpenModelica, except I just return the list of classes in the total model. And I run that on all of the models marked with experiment annotation. If it fails to get a total model, or example because the patched trunk does not have a replaceable GravityField, that experiment is silently discarded (but it should be very few that fail).

modelica-trac-importer, 2017-01-15:
**Comment by sjoelund&#x2e;se on 15 Jul 2014 19:13 UTC**
Replying to [comment:1 otter]:
> Partially fixed in eaa6a18f4ad45992a677af355846ae9624c98a10 by adding a model ModelicaTest.TestAllFunctions with an experiment annotation that calls most function tests.

Looking at that model, I think it would be better if the test functions were all different models. Because that gives you a better overview of what parts of MSL functions a tool supports.

Anyway, https://test.openmodelica.org/libraries/ModelicaTest_trunk/Coverage.txt and https://test.openmodelica.org/libraries/MSL_trunk/Coverage.txt are up now (refreshed daily, and you can see how many experiments use them; I could also provide a list of classes used in MSL by MSL,Modelica,Buildings,ThermoPower,etc libraries if desired).

beutlich, 2018-11-12:
> Looking at that model, I think it would be better if the test functions were all different models. Because that gives you a better overview of what parts of MSL functions a tool supports.

Resolved by c0262ce96d4832a64d48358f382219b48e613908 and 6173925548cad2f94f07e6c338597307d5e921a2 (for MSL v3.2.2).

beutlich, 2019-05-27:
@sjoelund Can you please sum up what is left to do here?

sjoelund, 2019-05-27:
The old way we used of generating those files seems to have disappeared, but I hacked up something similar. The following is a list of the classes not tested by any experiment in Modelica or ModelicaTest: https://gist.github.com/sjoelund/f5a215311f779783d8c192c0c0e66c6b

The script used to generate the list creates one big model containing all experiments, creates a total model of these and compares the classes present before and after: https://gist.github.com/sjoelund/7df830e1ea8be4b3032023518cf9542d

beutlich, 2019-05-27:
Oh dear, that is a huge number. 🔢

sjoelund, 2019-05-27:
It hasn't filtered out things like icons and documentation, but it's still somewhat large.

HansOlsson, 2019-05-28:
> Oh dear, that is a huge number

Yes, that was so troubling that I thought it best to have double-check that to be sure - and Dymola gave a similar result(*), and found another number class coverage is at **71.5%**.

Unfortunately that number is not huge.

*: Except that we don't include functions in this list yet; on the other hand it remove documentation and icons - except for Modelica.Fluid.Dissipation.Utilities.Icons, and it is laid out differently  - so I didn't upload it yet; and I'm not sure if it was for latest build. The important part was the top of the summary that says: 1567 of 2191 classes are covered by the test suite. 624 classes are not covered.

beutlich, 2020-03-14:
@HansOlsson @sjoelund Can you please regenerate the class coverage based on MSL v4.0.0-beta.2? Thanks.

HansOlsson, 2020-03-18:
I reran `ModelManagement.Check.checkLibrary(false, condCov=false, styleCheck=false, name="Modelica", testFileDirectory="ModelicaTest", filter=true, localCoverage=true, translationStructure=false);` on trunk. 
Which for our code required some conversions of additional libraries https://github.com/modelica/Modelica_LinearSystems2/pull/72

I have attached the log.

[Modelica_HTML_log.zip](https://github.com/modelica/ModelicaStandardLibrary/files/4349712/Modelica_HTML_log.zip)

HansOlsson, 2020-03-19:
I realized that I missed computing the simple statistics based on that. 

The class coverage is now 83.3% up from 79.3%.
The number of classes increased from 2191 to 2431, i.e. 240 new classes.
The covered classes increased from 1739 to 2024, i.e. 285 more.

beutlich, 2020-03-19:
Thanks a lot for the information.

beutlich, 2020-03-20:
> I have attached the log.

It's also available [here](https://gistpreview.github.io/?ef7741a7e8cc8949debbc684573d3086) directly in browser.

dietmarw, 2020-03-20:
@HansOlsson I was wondering if Dymola is able to output the coverage in a slightly different format simply listing the full class name and the amount of usage all in one line. 
E.g., instead of:
```
package Blocks
    package Examples
        model PID_Controller, used 1 time.
```
something like this:
```
Blocks.Examples.PID_Controller, used 1 time.
```
That kind of format would make it much easier to grep and filter and create a reduced (working) list.

dietmarw, 2020-03-20:
And simple text format would be sufficient for the post-processing application.

beutlich, 2020-03-20:
> 
> 
> And simple text format would be sufficient for the post-processing application.

Sorted by names or by number of occurences (as https://test.openmodelica.org/libraries/MSL_trunk/Coverage.txt does)? (By the way, that MSL_trunk in OpenModelica is not related to MSL v4.0.0.)

dietmarw, 2020-03-20:
Yes that would be perfect. @sjoelund Is there a more up to date listing available?

HansOlsson, 2020-03-20:
> @HansOlsson I was wondering if Dymola is able to output the coverage in a slightly different format simply listing the full class name and the amount of usage all in one line.

Not at the moment as far as I know. I will see if we can prioritize it.

sjoelund, 2020-03-26:
MSL_trunk in OpenModelica is the same as the latest master, but that particular file didn't get updates since 2018. I'll see how long it takes to regenerate these files...

sjoelund, 2020-03-26:
[count.txt](https://github.com/modelica/ModelicaStandardLibrary/files/4386988/count.txt)
I believe I could create a Jenkins pipeline to run that on new commits to the branches. The file itself is based on an 11 day old MSL since apparently that's what I latest recompiled on my machine.

beutlich, 2020-03-26:
> The file itself is based on an 11 day old MSL

Cannot believe it, since there no longer is Modelica.SIunits in MSL.

sjoelund, 2020-03-26:
Oh yeah, I forgot to change to the trunk version. Silly me.

sjoelund, 2020-03-26:
[CountClasses.txt](https://github.com/modelica/ModelicaStandardLibrary/files/4387278/CountClasses.txt)
from [Jenkins](https://test.openmodelica.org/jenkins/blue/organizations/jenkins/ModelicaAssociation%2FModelicaStandardLibrary/detail/PR-3524/2/artifacts/) for #3524 (link to the artifacts is removed when that PR closes). Merge that PR if you want this to be generated continuously (cherry-pick it to a maintenance branch if you want the information there as well).

beutlich, 2025-07-29:
> I reran `ModelManagement.Check.checkLibrary(false, condCov=false, styleCheck=false, name="Modelica", testFileDirectory="ModelicaTest", filter=true, localCoverage=true, translationStructure=false);` on trunk. Which for our code required some conversions of additional libraries [modelica/Modelica_LinearSystems2#72](https://github.com/modelica/Modelica_LinearSystems2/pull/72)

Coming from #4684 I reran that command on 8ae3d35c24e519cb2996cab20f3b13daf2b0c50a (MSL v4.1.0) in Dymola 2025x R1, yielding [this log output](https://gistpreview.github.io/?5d1f2f8be0e48d7eaedb8cc002338548).