modelica/ModelicaStandardLibrary #1627: Time table does not generate time events

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 10 Dec 2014 15:56 UTC**
As noted in https://trac.modelica.org/Modelica/ticket/1619#comment:8 the trunk version of time tables no longer generates time events in Dymola (and possibly other tools), due to the scaling of time. That decreases performance and accurancy.

The 3.3. Modelica Language Specification in 8.5 states that time>=... and time < ... may generate events. The previous version used time>=... which can generate time events, but the new version contains timeScaled>=...

I have not investigated exactly how to rewrite it.
It is less important than #1619, but should still be resolved before the library is released.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1627

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 10 Dec 2014 21:42 UTC**
In 70f01e3f87338629f5cffc7e739785d8226ae178 for #1473 I did not want to affect the C sources but only the Modelica sources, by purpose. I guess this needs to be edited (or reverted) according to your comment at comment 8 of #1619.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 5 Jan 2015 08:07 UTC**
Actually I cannot reproduce the problem in Dymola 2015 32-bit with current MSL trunk (829eaf386e14170688c87b6e1380957ab4e1ec04) and ModelicaStandardTables compiled with DEBUG_TIME_EVENTS set. If I run ModelicaTest.Tables.CombiTimeTable.Test67 I can see the expected event messages for each of the CombiTimeTable instances.
```
Log-file of program ./dymosim
(generated: Mon Jan 05 08:51:28 2015)

dymosim started
... "dsin.txt" loading (dymosim input file)
At time 0.000000: 1. time event at 400.000000
At time 0.000000: 1. time event at 6.666667
At time 0.000000: 1. time event at 0.111111
At time 0.000000: 1. time event at 0.004630
At time 0.000000: 1. time event at 400000.000000
... "Test67.mat" creating (simulation result file)

Integration started at T = 0 using integration method DASSL
(DAE multi-step solver (dassl/dasslrt of Petzold modified by Dynasim))
At time 400.000000: 2. time event at 2200.000000
At time 6.666667: 2. time event at 36.666667
At time 0.111111: 2. time event at 0.611111
At time 0.004630: 2. time event at 0.025463
At time 400000.000000: 2. time event at 2200000.000000
At time 2200.000000: 3. time event at 4000.000000
At time 36.666667: 3. time event at 66.666667
At time 0.611111: 3. time event at 1.111111
At time 0.025463: 3. time event at 0.046296
At time 2200000.000040: 3. time event at 4000000.000000
Integration terminated successfully at T = 4000
   CPU-time for integration      : 0.016 seconds
   CPU-time for one GRID interval: 0.032 milli-seconds
   Number of result points       : 505
   Number of GRID   points       : 501
   Number of (successful) steps  : 52
   Number of F-evaluations       : 100
   Number of H-evaluations       : 569
   Number of Jacobian-evaluations: 48
   Number of (model) time events : 0
   Number of (U) time events     : 0
   Number of state    events     : 2
   Number of step     events     : 0
   Minimum integration stepsize  : 0.008
   Maximum integration stepsize  : 1.05e+003
   Maximum integration order     : 1
Calling terminal section
No more time events for time > 4000.000000
No more time events for time > 66.666667
No more time events for time > 1.111111
No more time events for time > 0.046296
No more time events for time > 4000000.000000
... "dsfinal.txt" creating (final states)
```
These messages are generated by getNextTimeEvent = ModelicaStandardTables_CombiTimeTable_nextTimeEvent.
```mo
  when {timeScaled >= pre(nextTimeEventScaled),initial()} then
    nextTimeEventScaled = getNextTimeEvent(tableID, timeScaled, tableOnFileRead);
    // ...
  end when;
```

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 18 Mar 2015 16:43 UTC**
Hans, is this issue still relevant for you? You did not answer on my comment:2.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 10 Apr 2015 15:07 UTC**
Replying to [comment:3 beutlich]:
> Hans, is this issue still relevant for you? You did not answer on my comment:2.

At least on trunk the issue is still present, and this is still relevant for us.
(I have been busy I didn't have time to investigate the new version MSL earlier.)

You can also see it from the log, it has 2 state events and 0 model time events; it should be the other way.

The naïve solution would be:
```
  when {time >= pre(nextTimeEvent),initial()} then
```
However, floating point numbers don't always work as expected (unless timeScale is power of 2 - such as 1).

Here we will take 'nextTimeEventScaled' and ideally generate an event at:
'(nextTimeEventScaled\*timescale)/timeScale'.
This can differ from 'nextTimeEventScaled' - in particular it could be smaller than 'nextTimeEventScaled', i.e. generate a time-event too early.

modelica-trac-importer, 2017-01-15:
**Comment by pharman on 11 Apr 2015 08:07 UTC**
This is also an issue for the CyDesign compiler, since the scaling was added we no longer identify the time event.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 11 Apr 2015 20:48 UTC**
I see. What do you recommend?
 1. Revert 70f01e3f87338629f5cffc7e739785d8226ae178.
 2. Change from _when time/timeScale ≥ nextTime_ to _when time ≥ nextTime\*timeScale_.

modelica-trac-importer, 2017-01-15:
**Comment by pharman on 11 Apr 2015 20:52 UTC**
Replying to [comment:6 beutlich]:
> I see. What do you recommend?
>  1. Revert 70f01e3f87338629f5cffc7e739785d8226ae178.
>  2. Change from _when time/timeScale ≥ nextTime_ to _when time ≥ nextTime\*timeScale_.

I believe 2 would be a solution.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 13 Apr 2015 08:16 UTC**
Replying to [comment:7 pharman]:
> Replying to [comment:6 beutlich]:
> > I see. What do you recommend?
> >  1. Revert 70f01e3f87338629f5cffc7e739785d8226ae178.
> >  2. Change from _when time/timeScale ≥ nextTime_ to _when time ≥ nextTime\*timeScale_.
> 
> I believe 2 would be a solution.

The problem is that if timescale is a non-power of 2 it seems this could generate an event too early. If that occurs then `time>=pre(nextTime)*timescale` will be true at the event, i.e. it will NOT generate any new events after that point.

There are two ways of handling that:
1. Prove that the logic using isNearlyEqual avoids that. (Personally I don't really like adding epsilon everywhere (without careful analysis giving exact bounds) and would be happier without isNearlyEqual.)
2. Detect the rare case when the problem occurs and handle that. The rare case is that `...nextTimeEvent(...)` returns nextTimeEventScaled such that `nextTimeEventScaled*timeScale<=time` - in that case adding some suitable epsilon and calling `...nextTimeEvent` again should work; that could be hidden in an updated getNextTimeEvent (=handled on the Modelica-side without changing the C-library).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 13 Apr 2015 08:24 UTC**
Indeed, _Test67.t_min_ gives no time event at time=2200s.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 13 Apr 2015 10:23 UTC**
Resolved by be385e3be01f5e44562da98dd48bbf92fd37f714. Hans, Peter, can you please test and confirm. Thanks.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 15 Apr 2015 10:10 UTC**
Argh, wait. be385e3be01f5e44562da98dd48bbf92fd37f714 needs a further revision. Sorry for that.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 17 Apr 2015 07:21 UTC**
Detection of time events was fixed in bdbec63f67ea78999700293d22dbb73f2153086a (trunk) and fb6617f2b8c949dd0598799ffc1f29d781e70f34 (maint.).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 28 Sep 2015 11:38 UTC**
Tickets on not released features should be set to hidden in order to not appear in the release notes later.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 29 Oct 2015 15:19 UTC**
Do no longer hide development tickets. Instead leave the Release Notes empty.

modelica-trac-importer, 2017-01-15:
**Comment by jesse&#x2e;gohl on 4 Mar 2016 19:30 UTC**
I checked the following model with Dymola 2017 beta 1 and the MSL 3.2.2 that it ships with. 

```
model Unnamed
  Modelica.Blocks.Sources.CombiTimeTable combiTimeTable(table=[0.0,0; 1,1; 1.3,4; 1.5,0.1; 1.7,3; 2.3,0.2; 3,3; 4,2; 15,2]);
  Modelica.Blocks.Continuous.Integrator integrator(k=0.001);
equation 
  connect(combiTimeTable.y[1], integrator.u);
  annotation (uses(Modelica(version="3.2.2")));
end Unnamed;
```

It appears that events are not being generated.  Simulating with data stored at events and not equidistant time grid with the default solver settings I get the attached image for the output of the table and the following simulation log:

```
Integration terminated successfully at T = 4
CPU-time for integration : 0.001 seconds
CPU-time for one GRID interval: 0.0385 milli-seconds
Number of result points : 27
Number of GRID points : 27
Number of (successful) steps : 26
Number of F-evaluations : 70
Number of H-evaluations : 27
Number of Jacobian-evaluations: 15
Number of (model) time events : 0
Number of (U) time events : 0
Number of state events : 0
Number of step events : 0
Minimum integration stepsize : 0.004
Maximum integration stepsize : 0.512
Maximum integration order : 3
Calling terminal section
... "dsfinal.txt" creating (final states)
```
I think we would expect to see data points at 1, 1.3, 1.5, etc. so it appears that events are not being generated.
[image]

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 4 Mar 2016 19:35 UTC**
FWIW, this has caused irritation by a customer (using MSL 3.2.1 with the bug) because it led to wrong results.

modelica-trac-importer, 2017-01-15:
**Comment by jesse&#x2e;gohl on 4 Mar 2016 19:48 UTC**
Replying to [comment:15 jesse.gohl@…]:
> I checked the following model with Dymola 2017 beta 1 and the MSL 3.2.2 that it ships with. 
> ...

One additional comment, could there not be a parameter to allow the user to choose whether to generate events or not?

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 4 Mar 2016 20:20 UTC**
I now wonder if my understanding of "The table is implemented in a numerically sound way by generating time events at interval boundaries." is/was wrong since MSL 3.2.1. Please read again section 2.1 on time events of http://dx.doi.org/10.3384/ecp14096893 to understand the implemented design.
You need to replicate the sample points where you want to force the time events, i.e., in your model _Unnamed_ change the _table_ to `[0,0;0,0;1,1;1,1;1.3,4;1.3,4;1.5,0.1;1.5,0.1;1.7,3;1.7,3;2.3,0.2;2.3,0.2;3,3;3,3;4,2;4,2;15,2;15,2]`.
If this is really an issue of interpretation please close this ticket again and open a new ticket since it is a different topic.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 4 Mar 2016 21:25 UTC**
Improved documentation of CombiTimeTable in 04b24667fc5cd5450fb41a7cdf307da40577ed38:

_The table is implemented in a numerically sound way by generating time events at interval boundaries. An interval boundary is defined by two identical time values following each other. For example_

```
   table = [0  0
            1  0
            1  1
            2  4
            3  9
            3  2
            4  3];
```
_defines three intervalls: 0..1, 1..3, 3..4. Within an interval the defined interpolation method is applied (so the table outputs within an interval are continuous,and if the interpolation method is smooth, also continuously differentiable). No time events are generated within an interval in order that also intervals with many points do not reduce the simulation efficiency._

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 4 Mar 2016 21:39 UTC**
I am not sure that this actually is numerically sound with variable step integrators in all situations. The reason why we reported this as a bug is that a customer saw very unexpected results since the integrator took far too few steps to capture the table data even remotely correct. While I agree that the implementation now corresponds to the documentation, I don't agree, from practical experience, with the term "numerically sound". So I think the proposal above that allows to have a time event at every data point should be an option in the model, and it allows to check whether events at every data point can be skipped in a particular case. Let me check with my colleagues whether there is a chance to reproduce the error, since this happened on a proprietary model. I would not report this if this had not caused real-world trouble!

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 4 Mar 2016 22:06 UTC**
The following model gives vastly different results depending on the integrator and tolerance:

```
model Test
Modelica.Blocks.Sources.CombiTimeTable combiTimeTable(table=[0.0,0; 1,1; 1.3,4; 1.5,0.1; 1.7,3; 2.3,0.2; 3,3; 4,2; 15,2]);
Modelica.Blocks.Continuous.Integrator integrator(k=0.001);
equation 
connect(combiTimeTable.y[1], integrator.u);
annotation (uses(Modelica(version="3.2.2")));
end Test;
```
See the results in the attached plot.

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 4 Mar 2016 22:14 UTC**
The slow integrator could be a mass getting an external heat input in a physical model. This is really not numerically safe, and an option to generate events would cure this. If one uses TimeTable instead, which generates events, and the issues are gone. Don't get me wrong; I like the idea and option of being able to skip the events, but it is not always safe. I agree that it conforms to the documentation though, and I am fine to move this to a later release, but then we should be more explicit about risks in the documentation.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 4 Mar 2016 22:16 UTC**
I can see the real-world trouble. If I remember correctly the old Dymola-only tables implementation generated time events for each sample point. In the new tables implementation (since MSL 3.2.1) this was changed to only happen at repeated sample points. I did it by purpose ("It is guaranteed that there are no time events at interval boundaries with a simple sample point only.") but am no longer sure if it really is the intended design according to the documentation "... by generating time events at interval boundaries." - meaning at all interval boundaries. It looks like I merged the potential discontinuities and the time-events behavior to happen at the repeated sample points.

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 4 Mar 2016 22:20 UTC**
And of course the input to the integrator looks even worse, for the three cases.

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 4 Mar 2016 22:25 UTC**
The problem is that this depends so much on the use of the table in the model. If the integrator in my example is changed to a high gain (k=1000), the difference becomes small. My opinion is that the (safe) default should be to generate a time event at every data point, and the higher performance parameterized option should be to disable events except at double sample points.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 4 Mar 2016 22:26 UTC**
For me it is not a documentation issue. If CombiTimeTable should behave like TimeTable (which has the same documentation "The table is implemented in a numerically sound way by generating time events at interval boundaries") than the design decision taken back in 2013 for the open-source CombiTimeTable was wrong.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 4 Mar 2016 22:33 UTC**
No, the decision was correct. If you want to have time events at every point, duplicate the points and you get time events. Therefore, it is possible to model what you want to do.

Assume the table reads measurement data, say every 1ms. If a electro-mechanical model like a robot is simulated using this measured data with DASSL, the simulation time goes up for a factor of 10-50. 

Many users would be VERY UPSET, if a new MSL release suddenly makes their model extremely slow.

It is fine to discuss to add another option, or make a separate model that has time events at every table data point. However, EVERY CHANGE MUST BE BACKWARDS COMPATIBLE, if reasonably possible. And here, it is very simple. The default option should not change the previous behavior.

We are shortly before a release of MSL 3.2.2 and we cannot introduce new features now to MSL. Therefore, I schedule this ticket to a future release.

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 4 Mar 2016 22:42 UTC**
Martin, and also the people from Dassault: please note that the default behavior of Dymola before there was an MA implementation of the tables was to have time events, so this is a Change in behavior at least for Dymola users that don't update every version (It may have been this way in MSL. 3.2.1). And you make your decision based on one use-case that you care for, but there are many other ones. As is, the component is dangerous to use for anyone but real power users! If the behavior was without events in 3.2.1, I am fine to make that the default. If I use the same emphasis as you, I'd say. 

Many users would be VERY UPSET if they discover that using a model from MSL can lead to VERY LARGE errors. Which I demonstrated. The minimum task to move this to next_MINOR-version is to document the risk.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 4 Mar 2016 22:56 UTC**
I checked with Dymola 2016 (one year ago), and the behavior is the same as in Dymola 2017 Beta.1 (no time events are generated)

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 5 Mar 2016 00:01 UTC**
I suggest the following update/addition to the documentation, before releasing MSL 3.2.2: 

* Note that this behavior, though efficient, may lead to unexpected results and large deviations between different integrators and tolerance settings when large integrator steps lead to skipping of single data points from the table. This can happen when the output from the table does not influence dynamic states strongly, e.g. when that output is an input to a low gain integrator.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 5 Mar 2016 07:51 UTC**
Improved documentation in 8552be62ca5578ba90bdbf6396ac5fc85c7c3c78 (the underlined/italic text has been newly introduced):

The table is implemented in a numerically sound way by generating time events at interval boundaries. An interval boundary is defined by two identical time values following each other. For example

```
   table = [0, 0;
            1, 0;
            1, 1;
            2, 3;
            3, 5;
            3, 2;
            4, 4;
            5, 5];
```
defines three intervalls: 0..1, 1..3, 3..5. Within an interval the defined interpolation method is applied (so the table outputs within an interval are continuous,and if the interpolation method is smooth, also continuously differentiable). No time events are generated within an interval in order that also intervals with many points do not reduce the simulation efficiency** **'<ins>(note in package Modelica version 3.2 and earlier, time events had been generated).</ins>_
___If the table points are largely changing, it is adviseable to force time events by duplicating every time point (especially, if the model in which the table is present allows the variable step integrator to make large integrator steps). For example, if a sawtooth signal is defined with the table, it is more reliable to define the table as: ___

```
   table = [0, 0;
            1, 2;
            1, 2;
            2, 0;
            2, 0;
            3, 2;
            3, 2];
```
___instead of ___

```
   table = [0, 0;
            1, 2;
            2, 0;
            3, 2];
```
___because time events are then generated at every time point.___

For an upcoming MSL release, this behaviour should not be changed (because not backwards compatible). 

One might add an option (or make an additional table class), to generate an event at every table point. However, the question is whether this only holds for linear interpolation, or also for smooth interpolation (Smoothness = ContinuousDerivative, MonotoneContinuousDerivative1, MonotoneContinuousDerivative2). The latter would be surprising.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 5 Mar 2016 08:24 UTC**
The original ticket from Hans was totally different. Therefore I propose to add this proposed new option to either #1153 or make a new extra ticket for it. And close this ticket again for MSL 3.2.2.

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 6 Mar 2016 20:02 UTC**
Replying to [comment:31 otter]:
> Improved documentation in 8552be62ca5578ba90bdbf6396ac5fc85c7c3c78 (the underlined/italic text has been newly introduced):> ...
><ins>_If the table points are largely changing, it is adviseable to force time events by duplicating every time point (especially, if the model in which the table is present allows the variable step integrator to make large integrator steps). For example, if a sawtooth signal is defined with the table, it is more reliable to define the table as: _</ins>
> 
> ```
>    table = [0, 0;
>             1, 2;
>             1, 2;
>             2, 0;
>             2, 0;
>             3, 2;
>             3, 2];
> ```
><ins>_instead of _</ins>
> 
> ```
>    table = [0, 0;
>             1, 2;
>             2, 0;
>             3, 2];
> ```
><ins>_because time events are then generated at every time point._</ins>
> 
> For an upcoming MSL release, this behaviour should not be changed (because not backwards compatible). 
> 
> One might add an option (or make an additional table class), to generate an event at every table point. However, the question is whether this only holds for linear interpolation, or also for smooth interpolation (Smoothness = ContinuousDerivative, MonotoneContinuousDerivative1, MonotoneContinuousDerivative2). The latter would be surprising.

Let us stick with the linear interpolation first. I think a user can rightly expect that the integrator follows the trajectory of the input signal, for which I demonstrated that this is not the case. This is definitely against expectations, even  though sort of documented. Users that don't want the input trajectory to be followed at the sampling rate of the data should down-sample their data to the required accuracy (from 1 sample/ms to 1 sample/ 10 ms or to 1 sample per 50 ms? how should the tool guess). Your preferred solution forces users to to double sample points to get the expected behavior and it is completely backwards from my point of view. Your point of view is one of many use-cases, and I am in favor to support it through an option. Since we want to keep backward compatibility, I am also in favor to have it as default. 

Depending on the data, it may even be necessary for interpolation options to force steps to not deviate from the input data. With coarser data, forcing a time event at data points may be just right, but that probably requires a deeper investigation.

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 6 Mar 2016 20:03 UTC**
Replying to [comment:32 beutlich]:
> The original ticket from Hans was totally different. Therefore I propose to add this proposed new option to either #1153 or make a new extra ticket for it. And close this ticket again for MSL 3.2.2.

Hans' ticket and this discussion is both about the issue of time events in the tables,so I think it is related.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 8 Apr 2016 14:09 UTC**
This was discussed at the [attachment:160406_Web_meeting_MSL_Tables_development.pdf:ticket:1153 web meeting] with the following result:
Three options to solve
1. Keep the existing behavior and introduce a new interpolation choice that always generates events (at linear interpolation)
2. Fix the existing behavior (to always generate events) in order to match the TimeTable behavior. (There still is Clock fed into Combi1DTable as eventless  alternative/workaround)
3. Same as 2. but additionally introduce a new interpolation choice that behaves like the current implementation (only generate events at replicated sample points).
I'll soon set up a MAP-Lib poll which alternative is the preferred one.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 15 Apr 2016 11:41 UTC**
Replying to [comment:35 beutlich]:
> I'll soon set up a MAP-Lib poll which alternative is the preferred one.

Feel free to contribute to the decision finding on http://doodle.com/poll/9s7zhtnmguev5c8m before May 1, 2016. Thank you for collaboration.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 16 Apr 2016 14:39 UTC**
Replying to [comment:35 beutlich]:
> This was discussed at the [attachment:160406_Web_meeting_MSL_Tables_development.pdf:ticket:1153 web meeting] with the following result:
> Three options to solve
> 1. Keep the existing behavior and [...]

Please, explain why you think that 2./3. are an alternative?
I think it is very bad for users if in one MSL version we improve functionality (more efficient simulation) and in one of the follow-up versions we go back to the behavior of an older versions and add the functionality in a different way.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 9 Aug 2016 11:35 UTC**
Replying to [comment:37 otter]:
> Please, explain why you think that 2./3. are an alternative?
> I think it is very bad for users if in one MSL version we improve functionality (more efficient simulation) and in one of the follow-up versions we go back to the behavior of an older versions and add the functionality in a different way.

I can only repeat myself: see comment:26. TimeTable and CombiTmeTable have same documentation but different behaviour (w.r.t time events) which is not as expected.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 9 Aug 2016 11:39 UTC**
Voting results: 4 votes for "Keep existing behavior and introduce a new interpolation choice that always generates events" vs. 2 votes for "Fix existing behavior (to always generate events) in order to match the TimeTable behavior, but additionally introduce a new interpolation choice that behaves like the current implementation. (Still have Clock + Combi1DTable)"

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 27 Oct 2016 10:14 UTC**
See #2080.

modelica-trac-importer, 2017-01-15:
**Comment by Filip Jorissen on 27 Oct 2016 10:16 UTC**
I have had similar issues with tables. The proposed solution seems to be a decent fix.

However, in my opinion this is really an integrator issue. The integrator is supposed to make sure it achieves a certain tolerance, which it is failing to do in this case. I guess the problem is that the integrator uses some support points using which it iterates until convergence. These support points may not be sufficiently close together to capture the actual dynamics of the inputs of the problem. When the solver is able to satisfy its tolerance criterion it will simply accept the solution and continue to the next step. A solution may be to sample the inputs of the problem (all variables that are computed as a direct function of 'time') and to see how well the inputs can be approximated using a linear/quadratic/whatever function of time. (depending on what the integrator assumes) If there are fast/important dynamics in the inputs then this may be used as a metric to set an upper bound for the integrator step size.

An example that illustrates this, without the use of tables, is the following model:

```mo
model test
  Real y;
equation 
  der(y)= 1+100*abs(sin(time/1000)^1000);
  annotation ();
end test;
```

This integrates a (smooth) function/equation that is usually equal to 1 but contains a peak value of 100 every few hundred seconds. If you integrate this (in Dymola 2017) with DASSL or LSodar (tolerance = 1e-4) and Euler (delta t= 100s) then you get the figure I uploaded. Euler returns the 'correct' result. Dassl and LSodar greatly underestimate the result. Dassl output fyi:

Integration terminated successfully at T = 1000000
   CPU-time for integration      : 0.00732 seconds
   CPU-time for one GRID interval: 0.0146 milli-seconds
   Number of result points       : 501
   Number of GRID   points       : 501
   Number of (successful) steps  : 44
   Number of F-evaluations       : 101
   Number of Jacobian-evaluations: 47
   Number of (model) time events : 0
   Number of (U) time events     : 0
   Number of state    events     : 0
   Number of step     events     : 0
   Minimum integration stepsize  : 5e-05
   Maximum integration stepsize  : 2.88e+05
   Maximum integration order     : 1

You may argue that this is a stupid example, but it illustrates the problem.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 27 Oct 2016 10:32 UTC**
It is unlikely that an algorithm is able to handle such problems satisfactory, without having additional specific information in the model (an algorithm computes the model at particular time instances and if at these time instances the fast change of a model is not visible in the model value or its derivative, there is no chance that the algorithm can handle this). A typical pracical example are mechanical impacts (say with elastic contact description). In order to improve the simulation accuracy, the modeler might trigger a state event once the impact boundary is reached.

As a side note: The Modelica tools typically transform to ODE form. This means that step-size control is ONLY on state variables (variables appearing in der(..)). If other variables are changing quickly (e.g. inputs), the user must give additional information (e.g. trigger events, or if the tool allows it define a maximum step size).

beutlich, 2017-04-18:
:tada: Finally closing another long-standing and heavily commented issue on the external tables.

closed 2017-04-18 by beutlich