modelica/ModelicaStandardLibrary #1666: RealOutput in SM_PermanentMagnet, etc. show up in component icon layer

modelica-trac-importer, 2017-01-15:
**Reported by jriel on 25 Feb 2015 18:43 UTC**
In 
```
Modelica.Electrical.Machines.BasicMachines.SynchronousInductionMachines.SM_ElectricalExcited
Modelica.Electrical.Machines.BasicMachines.SynchronousInductionMachines.SM_PermanentMagnet
Modelica.Electrical.Machines.BasicMachines.SynchronousInductionMachines.ReluctanceRotor
```
the **ir** and **idq_dr** ports, which are only used internally, are declared as **public**.  Because they use Modelica.Blocks.Interfaces.RealOutput, which has an associated icon annotation, an icon for these ports appears in the component icon.

I believe that these ports should be **protected**.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1666

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 19 Mar 2015 09:38 UTC**
See e6a020a43dfe822d11a03347bf4886c0d5166af5 for #1223.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 4 May 2015 11:08 UTC**
Screenshot from SimulationX: [image]

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 4 May 2015 13:06 UTC**
To me this looks like a problem with the non-definition of what is supposed to be displayed when no `Placement` annotation is provided. As I could see Dymola and OMEdit both decide to simply not display the graphical annotation when no `Placement` is given. SimulationX and Maplesim seem to display it in the component origin (if I understand it correctly).  But I could not find any mention of how to handle this in the Spec. So none of the tools is probably wrong or right.

Making the component protected would be a workaround but the underlying question of how components w/o `Placement` annotations should be displayed should also be clarified.

modelica-trac-importer, 2017-01-15:
**Comment by ahaumer on 22 Jun 2015 15:10 UTC**
IMHO this is a tool issue: shall an "input" without graphical annotation be displayed? 

We have to use the "input" prefix to be able to set the variables later, extending from 
```
Modelica.Electrical.Machines.Interfaces.PartialBasicInductionMachine
```

Making these variables protected prevents the user to access them for plotting, postprocessing, defining initial conditions, etc.

For now I don't find a solution -> postponed / shifted.

modelica-trac-importer, 2017-01-15:
**Modified by christiankral on 2 Nov 2015 06:58 UTC**

christiankral, 2017-07-05:
I agree with @AHaumer, this issue has to be treated in the Modelica Specification.

HansOlsson, 2017-07-06:
Since Modelica Language Specification 3.0 we have "visible" attribute as part of the Placement - together with the transformations. 

That will make it invisible from both icon and diagram-layer and as far I can see that was the intention in these cases.

christiankral, 2017-07-07:
This issue also applies to ``Modelica.Blocks.Interfaces.RealOutput idq_dr[2]`` of the machine models in ``Modelica.Electrical.Machines.BasicMachines.SynchronousInductionMachines``. Additionally,  the same issue applies to ``Modelica.Magnetic.FundamentalWave.BasicMachines.SynchronousInductionMachines`` and ``Modelica.Magnetic.QuasiStatic.FundamentalWave.BasicMachines.SynchronousMachines``.

closed 2017-07-07 by christiankral