modelica/ModelicaStandardLibrary #1724: Modelica.Blocks.Math.LinearDependency has strange formulation

modelica-trac-importer, 2017-01-15:
**Reported by Joe Riel on 1 Jun 2015 14:58 UTC**
The relatively new block LinearDependency seems strange. It's defining equation is `y = y0*(1 + k1*u1 + k2*u2);`. y0 is a parameter; its description says it is an initial value, which makes no sense.  It is a scale factor (with a useless default of 0).  However, if its value is nonzero, then one cannot produce an output with a non-zero offset.  I'm thinking the block would be slightly more useful if the defining equation were `y = y0 + k1*u1 + k2*u2`.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1724

modelica-trac-importer, 2017-01-15:
**Comment by Joe Riel on 1 Jun 2015 15:32 UTC**
A correction.  A nonzero value of y0, which is practically necessary to have a useful block, means the offset must be nonzero.   With the proposed change, the offset is independent of the gains.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 31 Aug 2015 09:29 UTC**
There is only one component of LinearDependency in the MSL, however the Real output Modelica.Blocks.Examples.RealNetwork1.linearDependency1.y is not used in the example model.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 28 Sep 2015 11:45 UTC**
The proposal would lead to a non-backwards compatible change of the library (which is not possible with MSL 3.2.1 build 4). Therefore, the milestone is moved to 3.2.2.

To avoid a non-backwards compatible change in 3.2.2, I would suggest to just improve the documentation of this block (and the use case for which it was designed).

modelica-trac-importer, 2017-01-15:
**Comment by ahaumer on 18 Dec 2015 12:05 UTC**
IMHO the implementation is erroneous and will be changed when a conversion script is applicable. This is stated in the documenation of the block.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 20 Dec 2015 20:45 UTC**
I reverted the changes from 811c96eb7a71c641c5896f9b27bbf34016f1d097 of Math.mo with 8bd005ca83368db93f4b1ed5737c0d82fd24d6d4 since

 * it is not clear if this change went in by design or by side-effect (no commit message was given)
 * the comment used the undesired h3 headline
 * the formula in the comment was invalid.

Please revise again for MSL 3.2.2. For that reason I changed the milestone back to MSL 3.2.2 which should be reset again once the comment is without errors. Sorry for the extra work.

modelica-trac-importer, 2017-01-15:
**Comment by ahaumer on 23 Dec 2015 18:28 UTC**
Please, do not revert - give a comment what's wrong in your opinion.

Once again: IMHO the implementation is wrong and will be corrected in the next major version (conversion script needed; ticket remains open). This is stated in the documentation (92389bffefad8506bd3854bebf16a843f89cf2df).

beutlich, 2017-01-27:
Point to @JoeRiel.

beutlich, 2019-08-09:
My proposal #2886 was rejected by @MartinOtter .

@AHaumer @MartinOtter Please provide an alternative fix!

JoeRiel, 2019-08-09:
On Fri, 09 Aug 2019 06:36:14 -0700
Thomas Beutlich <notifications@github.com> wrote:

> My proposal #2886 was rejected by @MartinOtter .
> 
> @AHaumer @MartinOtter Please provide an alternative fix!
> 

Hmm.  The original is definitely wrong and completely non intuitive.  
I understand the reluctance to change a model, as doing so will definitely
change any models that use it, but suspect it was never/rarely used
given its weirdness.   Maybe name the fixed model LinearDependent, or
TwoInputLinear, or something like that.

-- 
Joe Riel

beutlich, 2019-08-10:
#3088 is my second attempt to fix this issue.

commit 92389bff (2015-12-23): resolves #1724

git-svn-id: https://svn.modelica.org/projects/Modelica/trunk@8863 7ce873d0-865f-4ce7-a662-4bb36ea78beb
PR #3095 (2019-09-02): Fix formula of LinearDependency and convert parameters
This is my third attempt to fix #1724 superseding #2886 and #3088 after @JoeRiel's proposal to convert the parameters for the sake of backward compatibility.
closed 2019-09-02 by beutlich