modelica/ModelicaStandardLibrary #1747: HTML conventions are not followed for strong/emphasis

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 16 Jul 2015 10:48 UTC**
I thought I created this ticket a few minutes ago, but...

This is due to #1662, see also #1746

Modelica.UsersGuide.Conventions.Documentation.Format state that:
"For setting text in strong font (normally interpreted as boldface) the tags <strong> and </strong> have to be used.
For emphasizing text fragments <em> and </em> has to be used."

I understand the underlying reason and it is similar to the HTML-recommendation of not using them, http://www.w3.org/TR/html/text-level-semantics.html#the-b-element

However, searching in MSL 3.2r3 RC3:
1175 classes use <b>, but only 9 classes use <strong>
(and similarly for <i> vs <em>).

Either the conventions should be updated (e.g. removed - or just weakened considerably), or the rest of the MSL should be updated. 

Note that we SHOULD NOT just do a search-replace, since e.g. syntax highlighting should preferably replace <b> by <span> not by <strong>.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1747

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 25 Dec 2016 12:06 UTC**

beutlich, 2017-04-18:
Resolved in master.

closed 2017-04-18 by beutlich