modelica/ModelicaStandardLibrary #1752: Modelica.Electrical.Machines.Losses.InductionMachines.PermanentMagnetLosses needs connection for stator current

modelica-trac-importer, 2017-01-15:
**Reported by Joe Riel on 1 Aug 2015 20:50 UTC**
The Modelica.Electrical.Machines.Losses.InductionMachines.PermanentMagnetLosses has an input, is[m].  That should be replaced with either a real connector, or an electrical port.  As it is, it is not possible to use this model in a GUI because there is no port for the stator current.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1752

modelica-trac-importer, 2017-01-15:
**Modified by christiankral on 5 Aug 2015 20:11 UTC**

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 18 Aug 2015 12:06 UTC**

beutlich, 2017-01-27:
Point to @JoeRiel.

AHaumer, 2017-03-04:
Ok we need the the stator current(s) to caclulate the PM losses. 
There are 3 possibilites to implement that:
1. Use physical current connectors: Looks ugly and confusing
2. Use a current sensor and connect to a signal input of PM losses: Looks extermely ugly to use a sensor in a physical model!
3. Use an input without visible connector (legal Modelica!): Defining the input is possbile - why do I have to connect grafically?
To keep backwards compatibility, it is impossible to change the implementation.

beutlich, 2019-10-05:
>     3\. To keep backwards compatibility, it is impossible to change the implementation.

Backwards compatibility is not an issue for MSL v4.0.0 if a conversion can be provided.

@christiankral @AHaumer Do you think you can fix it for MSL v4.0.0? If not, please remove the milestone.

christiankral, 2020-02-15:
For the component `Modelica.Magnetic.FluxTubes.Basic.LeakageWithCoefficient` the `input` was replaced by a signal input without loosing backwards compatibility, see #3300, #3307 and #3311. 

@AHaumer we could still implement a similar solution this issue. Any comment?