modelica/ModelicaStandardLibrary #1761: Circular reference in Media

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 17 Aug 2015 10:27 UTC**
The last paragraph in Hubertus' commit message of 1e40f35e204bdd448ef3b2a09e83708f3bbb33d8 (for #799) reads:

The last move introduces a reference to Modelica.Fluid in Modelica.Media, which is a sort of circular reference. This will be flagged in a seprate ticket, and the models should be moved out out of Modelica.Media to MedelicaTest. The problem is that this _may_ be considered a non-backwards compatible change. I don't think so for models which only had the purpose to be test models.

Was this ever addressed? #1679 looks like a similar issue that was considered for build.3.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1761

modelica-trac-importer, 2017-01-15:
**Modified by dietmarw on 17 Aug 2015 12:07 UTC**

beutlich, 2020-06-07:
Resolved by #2996.

closed 2020-06-07 by beutlich