modelica/ModelicaStandardLibrary #1771: Wrong combiTimeTable output when startTime < first table point

modelica-trac-importer, 2017-01-15:
**Reported by filip&#x2e;jorissen on 1 Sep 2015 14:10 UTC**
When combiTimeTable.startTime is smaller than the first timestamp of a table then the wrong output values are returned. Consider the attached example, which produces the desired results and then set combiTimeTable.startTime to -1e7. Results are not the same for these two cases.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1771

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 1 Sep 2015 14:52 UTC**
If combiTimeTable.startTime is -1e7 then combiTimeTable.y![1] extrapolates to constant 1 since 1e7 is greater than combiTimeTable.t_max = 2049095. This is the expected behaviour, right?

modelica-trac-importer, 2017-01-15:
**Comment by filip&#46;jorissen on 1 Sep 2015 15:36 UTC**
It seems I misinterpreted the implementation of the model. 'startTime' has description "Output = offset for time < startTime". This does not seem to be correct? It's rather something like "Output = table(time-startTime)", i.e. it shifts all outputs in time. Based on this I was able to sort out the bug I was experiencing. If my interpretation is correct I suggest to adjust the description of 'startTime' and possibly rename the parameter too.

Filip

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 1 Sep 2015 18:24 UTC**
It's rather like "output = offset + if time < startTime then 0 else table(time-startTime)". I guess you did expect "output = offset + if time < startTime then 0 else table(time)". Thus indeed the table is shifted in time. At least it is documented.

Most of the source blocks have this parameter startTime, thus it is not simple thing to change.

beutlich, 2017-04-05:
@Mathadon I was working on #2190 (and assembled a solution utilizing CombiTimeTable) until I noticed that both IntegerTable and BooleanTable are based on absolute time, whereas CombiTimeTable is relative to startTime (with 0 as default). 😕 Thus my solution for #2190 only works for non-negative simulation start times right now. Therefore, I want to introduce a new public Boolean parameter to CombiTimeTable which ignores the startTime shifting. What name would you recommend: ignoreStartTimeShift, shiftByStartTime, startTimeShift? It then is `y = offset + if time < startTime then 0 else table(time - startTime)` if ignoreStartTimeShift is false and `y = offset + if time < startTime then 0 else table(time)` if ignoreStartTimeShift is true.

Mathadon, 2017-04-05:
I see you guys migrated to GitHub, awesome!

Your question makes me wonder what the use of `startTime` is in the first place? I see that it pops up in several blocks but I'd rather create a dedicated SISO block (`y =  if time > startTime then u else offset`) for such functionality rather than integrating it into the Tables (or `Ramp`, `Sine`, ...) blocks.

Wouldn't it also make sense that `CombiTimeTable`, `IntegerTable` and `BooleanTable` have the same interface? I.e. either all or none of them have `startTime`. I'm not sure if you that would avoid the need for your new parameter?

In `y = offset + if time < startTime then 0 else table(time - startTime)` you use the same parameter `startTime` for two distinct functionalities, which is something I generally try to avoid. Wouldn't it make sense to introduce a second parameter `parameter SI.Time timeDelay = startTime ` or something. This would avoid the need for the new boolean parameter?

In response to your question: I'd suggest `useStartTimeShift`. I think this is more easily interpretable than the other options :)

Sorry if I leave you with more new questions than answers!

beutlich, 2017-04-06:
Thanks @Mathadon for your fast reply. You are totally right about your thoughts

- BooleanTable, IntegerTable and CombiTimeTable (="RealTable" if you like) should have the same options regarding start time and time shift
- Using startTime also for the shifting is the root of the trouble you were facing and I am facing now. I believe, adding a new Parameter for the shifting (with default value set to startTime for backward comaptibility) can resolve the problem. This way you can set start time and time shift indepently, which is what you requested and we need,

beutlich, 2017-04-07:
Resolved in master branch.

closed 2017-04-07 by beutlich