modelica/ModelicaStandardLibrary #1773: Possible unspecified/arbitrary initial system

modelica-trac-importer, 2017-01-15:
**Reported by johan_ylikiiskila on 3 Sep 2015 11:46 UTC**
I am working with some Modelica.Magnetic models while using JModelica.org and discovered some strange behaviour in the model _Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SMPM_Inverter_MultiPhase_.

I was trying to reproduce the behaviour of the regression data I found at https://svn.modelica.org/projects/RegressionTesting/ReferenceResults/MSL/trunk when i found that JModelica.org started at a different initial point. I chose to investigate further and foudn the following behaviour where I need help:

The Two states _smpmM.stator.electroMagneticConverter.singlePhaseElectroMagneticConverter[1].Phi.re_ and _smpmM.stator.electroMagneticConverter.singlePhaseElectroMagneticConverter[2].Phi.im_ are initialized differently. In the reference data, the **imaginary part** is initialized to zero while the JModelica.org result has the **real part** initialized to zero. I started looking though the code to find where the initial conditions are set and found no start modifiers for the two states so I assume some default start values are used? 

To avoid the tool adding starting conditions, I tried adding a start modifier to the imaginary part I was able to make JModleica.org behave as the regression data.

To my question:
Is the initial system underdetermined so the different tools choose different starting conditions to add or is the starting state of the model defined and the tool (JModelica.org) is unable to determine that?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1773

christiankral, 2017-07-05:
The initialization of the example fully determined

### Mechanical Conditions ###

The mechanical initialization is performed in ``Modelica.Magnetic.FundamentalWave.Interfaces.PartialBasicInductionMachine`` by means of

```modelica
  output Modelica.SIunits.Angle phiMechanical(start=0) = flange.phi -
    internalSupport.phi "Mechanical angle of rotor against stator";
  output Modelica.SIunits.AngularVelocity wMechanical(start=0, displayUnit="rev/min")
    = der(phiMechanical) "Mechanical angular velocity of rotor against stator";
```

### Rotor Winding Magnetic Voltage ###

In ``Modelica.Magnetic.FundamentalWave.BasicMachines.SynchronousInductionMachines.SM_PermanentMagnet`` the rotor currents are initialized by 
```modelica
  Modelica.Blocks.Interfaces.RealOutput ir[2](
    start=zeros(2),
    each final quantity="ElectricCurrent",
    each final unit="A") if useDamperCage "Damper cage currents";
```
Consequently, the magnetic voltage of the rotor cage is defined to be zero.

### Stator Winding Magnetic Voltage ###

The stator currents are initialized in the example ``Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SMPM_Inverter_MultiPhase`` by 
```modelica
initial equation 
  smpm3.is[1:2] = zeros(2);
  smpmM.is[1:2] = zeros(2);
```
Thus, the magnetic voltage of the stator winding is equal to zero.

### Magnetic Circuit ###

For ``time = 0`` the only magnetic voltage of the magnetic circuit of the machine is the permanent magnet, ``smpmM.permanentMagnet.V_m.re = 505.528`` (A) and ``smpmM.permanentMagnet.V_m.im = 0``. So there occurs only the real part of the magnetic voltage. The total magnetic reluctance of the machine (in either axis) is equal to ``smpmM.airGap.R_m.d = smpmM.airGap.R_m.q = 1570.8`` (A/Vs). The total magnetic voltage is related with the magnetic flux by
```modelica
  (pi/2.0)*V_m.re = Phi.re*R_m.d;
  (pi/2.0)*V_m.im = Phi.im*R_m.q;
```
which is coded in ``Modelica.Magnetic.FundamentalWave.BasicMachines.Components.RotorSaliencyAirGap``.

These equations case the magnetic flux to be ``Phi.re = -0.505528`` (Vs) and ``Phi.re = 0``. Therefore, the magnetic flux is solely a consequence of the magnetic voltages (determined by the initial conditions) and the magnetic resistance of the magnetic circuit.

christiankral, 2017-07-05:
In the example ``SMPM_Inverter_MultiPhase`` the machine model are instantiated by the modifier  
```modelica
  Modelica.Magnetic.FundamentalWave.BasicMachines.SynchronousInductionMachines.SM_PermanentMagnet
    ...
    phiMechanical(fixed=true),
    wMechanical(fixed=true),
   ...
    ir(fixed=true));
```
When testing the example in JModelica I also noticed the following warning
```
Warning in flattened model:
  Iteration variable "smpm3.airGap.V_mrr.im" is missing start value!
Warning in flattened model:
  Iteration variable "smpm3.airGap.V_mrr.re" is missing start value!
```
which seems not to be correct according to my above explanation. 

However, JModelica reported a semantic error due to the missing ``each`` when assigning ``ir``. This issue shall be fixed in #2274.