modelica/ModelicaStandardLibrary #1774: dgeev and dgeev_eigenValues inconsistent

modelica-trac-importer, 2017-01-15:
**Reported by peter&#x2e;nilsson on 7 Sep 2015 08:04 UTC**
Modelica.Math.Matrices.LAPACK.dgeev uses Integer n=size(A, 1); as the third parameter while Modelica.Math.Matrices.LAPACK.dgeev_eigenValues uses size(A, 1). The former leads to int \* and the latter to const int \* input to Lapack function dgeev for the Dymola-generated prototype. Hence, it is impossible to be declare a Lapack prototype that is consistent with both. For gcc, it is even an error to have such a mismatching prototype.
Suggestion: Use Integer n=size(A, 1); also for dgeev_eigenValues.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1774

modelica-trac-importer, 2017-01-15:
**Comment by peter&#46;lundberg on 23 May 2016 08:53 UTC**
Also the 11th argument must be modifed to synchronized with dgeev:

I.e. change:

```
...
size(EigenvectorsL, 1),
work,
size(work, 1),
info) annotation (Library="lapack");
```

to

```
Integer ne=size(EigenvectorsL, 1);

...
ne,
work,
size(work, 1),
info) annotation (Library="lapack");
```

beutlich, 2017-02-06:
Resolved in master and maint/3.2.2.

closed 2017-02-06 by beutlich