modelica/ModelicaStandardLibrary #1778: No model of simple Coulomb friction

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 9 Sep 2015 12:12 UTC**
Currently there is no simple model of ideal dry fiction in Mechanics.Translational nor Mechanics.Rotational.

See the model Tutorial2015.BaseComponents.Friction.IdealDryFriction of [Dirk Zimmer's Modelica Tutorial](https://github.com/dzimmer/ZimmersModelicaTutorial) for a prototype implementation (lacking documentation) based on the already available Coloumb friction model of Mechanics.Translational.Interfaces.PartialFriction. Such a simple model with only the sticking and sliding force as constant parameters cannot be modeled by Mechanics.Translational.Components.SupportFriction, Mechanics.Translational.Components.Brake or Mechanics.Translational.Components.MassWithStopAndFriction currently.

It should be discussed if constant parameters are suitable for all modeling cases or if they should be replaced by inputs.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1778

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 23 Sep 2015 20:02 UTC**
From a discussion with dzimmer: The default value of v_small needs to be discussed. The rotational default value PartialFriction.w_small is set to 1e10 which is explained by #1493 and #701. The translational default value PartialFriction.v_small is set to 1e-3. It is not obvious why both values differ by such large order of magnitudes. For the new prototype IdealDryFriction I set v_small to final keeping its default value only to hide it from the tutorial users.

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 7 Nov 2016 21:51 UTC**

beutlich, 2017-08-09:
@MartinOtter Do you want a pull request for this enhancement? Or shall we simply close it as wontfix?

beutlich, 2019-12-21:
Can use MassWithStopAndFriction where Coulomb friction can be modelled.

closed 2019-12-21 by beutlich