modelica/ModelicaStandardLibrary #1783: Modification of final component in Modelica.Electrical.Machines.Interfaces.PartialBasicInductionMachine

modelica-trac-importer, 2017-01-15:
**Reported by jfrenkel on 11 Sep 2015 09:25 UTC**
In the model Modelica.Electrical.Machines.Interfaces.PartialBasicInductionMachine the component 
```
Machines.Losses.InductionMachines.Core statorCore(
	final coreParameters=statorCoreParameters
```
is modified. But in the Type of `stateCore` the component `m` of the component `coreParameters` is marked with final.

```
parameter Machines.Losses.CoreParameters coreParameters(final m=m);
```

Hence a modification of `coreParameters.m` is with respect to spec 7.2.6 not valid.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1783

modelica-trac-importer, 2017-01-15:
**Modified by jfrenkel on 11 Sep 2015 10:42 UTC**

modelica-trac-importer, 2017-01-15:
**Comment by ahaumer on 27 Sep 2015 17:06 UTC**
fixed by d416266f1a42047d2aa9a859038590754c050c71, f7b058c1d03887ed5ccd4a6f9198c154213b86ce

modelica-trac-importer, 2017-01-15:
**Changelog modified by beutlich on 29 Sep 2015 20:58 UTC**
Fixed modification of final component in Modelica.Electrical.Machines.Interfaces.PartialBasicInductionMachine

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 30 Sep 2015 09:04 UTC**
Also resolved by ffe95f04e12ffd091531f238a55425b901780a3d (Maint.) and 4e49e4e399870c7c6587f3b1ccb9859daad57537 (trunk) for PartialBasicDCMachine

modelica-trac-importer, 2017-01-15:
**Changelog modified by beutlich on 30 Sep 2015 09:04 UTC**
Fixed modification of final component in PartialBasicInductionMachine and PartialBasicDCMachine

closed 2017-01-15 by modelica-trac-importer

HansOlsson, 2022-03-17:
Since the change was reverted there seems to be an issue - note that Dymola 2022x only had logic to detect other cases of final-issues, not these one.

However, it is not clear whether the specification is ideal for this case.
What we have is a record class stating `final m=1`. Clearly replacing it with `m=2` would be bad even if done by modifying the entire record, but replacing it with `someRecord.m` where `someRecord.m` is of the same record class and thus also `1` isn't really a problem. 

How to specify that rule isn't clear, but it may indicate that the rules for overriding final modifier should be improved instead of trying to find a work-around in the library.

commit f7b058c1 (2015-09-27): fixes #1783

git-svn-id: https://svn.modelica.org/projects/Modelica/trunk@8492 7ce873d0-865f-4ce7-a662-4bb36ea78beb
commit d416266f (2015-09-27): fixes #1783

git-svn-id: https://svn.modelica.org/projects/Modelica/branches/maintenance/3.2.1@8492 7ce873d0-865f-4ce7-a662-4bb36ea78beb