modelica/ModelicaStandardLibrary #1791: comment string of M_Oline.segment references non-existent G_ref

modelica-trac-importer, 2017-01-15:
**Reported by Joe Riel on 2 Oct 2015 18:03 UTC**
The comment string of Modelica.Electrical.Analog.Lines.M_OLine.segment
includes "G_actual = G_ref/(1 + alpha\*(T_heatPort - T_ref))". 

There is no G_ref parameter; it should be G.  Similar errors occur in the M_OLine and OLine components, though they should probably refer to g rather than G (as R s should be r).

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1791

modelica-trac-importer, 2017-01-15:
**Comment by otter on 13 Dec 2015 18:24 UTC**
Since Kristin is in vacation, I propose to move this to the next minor milestone (it is anyway only a tiny issue with the documentation)

beutlich, 2017-01-27:
Point to @JoeRiel.

beutlich, 2017-02-01:
@kristinmajetta Can you please fix it. Thanks a lot.

kristinmajetta, 2017-04-04:
changed G_ref int G in models
Modelica.Electrical.Analog.Lines.OLine
Modelica.Electrical.Analog.Lines.M_Oline
Modelica.Electrical.Analog.Lines.M_Oline.segment

SHA-1: 70bc89a93c1808a078c3509bba50717319bb7310

beutlich, 2017-04-04:
Thanks for the fix.

beutlich, 2017-04-04:
Reopen for the sake of inconsistent parameter case. [Modelica.UsersGuide.Conventions.ModelicaCode.Naming](http://doc.modelica.org/help/Modelica_UsersGuide_Conventions_ModelicaCode.html#Modelica.UsersGuide.Conventions.ModelicaCode.Naming) says that capital G should be used for conductance, capital R for resistance. But model uses lower-case g and r as parameters and G and R in description formulars.

kristinmajetta, 2017-08-29:
changed R into R_actual, R_ref into R and G_ref into G in models
Modelica.Electrical.Analog.Lines.OLine
Modelica.Electrical.Analog.Lines.MOLine
Modelica.Electrical.Analog.Lines.ULine

SHA-1: e1d8e9020b5cd55b6b2c23fa625d8bc42a6b529b

kristinmajetta, 2017-08-29:
The models use lower-case g an r since those parameters refer to a resistance respectively to a conductance which is related to one meter of the line and not to the overal resistance or conductance of the line. Those overall resitance and conductance of the lines are calculated inside the line models. So the parameter names had been chosen to be lower-case to easily distinguish between the resistances and conductances PER meter and the OVERALL resistances and conductances.
This is why I would prefer to keep the lower-case r and g.

closed 2018-05-04 by kristinmajetta