modelica/ModelicaStandardLibrary #1808: M_OLine: default resistance/meter is too high

modelica-trac-importer, 2017-01-15:
**Reported by Joe Riel on 26 Oct 2015 17:17 UTC**
The default series resistance values for the Electrical.Analog.Lines.M_OLine component are supposed to be "sensible" for a 0.1 m microelectronic line, however, they appear to be several orders of magnitude too large (~ 0.5 MOhm/meter).

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1808

modelica-trac-importer, 2017-01-15:
**Comment by Joe Riel on 26 Oct 2015 18:51 UTC**
The documentation for the part states that it has coupled inductors "like in the M_Inductor" model.  That should be the "M_Transformer" model.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 13 Dec 2015 18:17 UTC**
If a default value should be changed, this is a non-backwards compatible change and requires a conversion script. Since MSL 3.2.2 is backwards compatible, I changed the milestone.

beutlich, 2017-01-27:
Point to @JoeRiel.

kristinmajetta, 2017-04-04:
changed "M_Inductor" into "M_Transformer" in the documenation of Modelica.Electrical.Analog.Lines.M_OLine

SHA-1: dbb447faabd150d781c7d8bb6812048d2dc88819

beutlich, 2019-11-02:
@JoeRiel Do yo think you could provide a pull request? I can take care of the conversion and test afterwards. Thanks you.

dietmarw, 2020-01-24:
Somone with a background in electronics should have a look at this. The only rough value for microeleectronic lines I could find is that they typically have a [sheet resistance](https://en.wikipedia.org/wiki/Sheet_resistance)  of about 0.5 mOhm/sq. But there are more issues with that model. In the documentation a principal picture is shown with values but those are displayed a absolute values in the image when in the models they are all relative values per meter. At the same time the example is supposed to represent a lenght of 0.1 m. So it really does not make sense and should be reworked. But this will most likely not be feasible to finish in time for MSL 4.0.0

JoeRiel, 2020-01-24:
0.5 mOhm/sq is the nominal value for 1 ounce copper, a typical thickness.  For PCB construction in the US, copper thickness is specified in ounces,  K ounce copper is equivalent to the thickness achieved by a copper sheet 1 foot square weighing K ounces.