modelica/ModelicaStandardLibrary #1822: Why is the Complex library Protected?

modelica-trac-importer, 2017-01-15:
**Reported by stefanv on 5 Nov 2015 17:50 UTC**
Why does Complex have a `Protection(access=Access.hide)` annotation? What's the big secret? This annotation prevents even the documentation from being visible.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1822

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 6 Nov 2015 07:31 UTC**
I think the intention is to hide it from the package browser view if nothing else.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 6 Nov 2015 08:54 UTC**
Replying to [ticket:1822 stefanv]:
> Why does Complex have a `Protection(access=Access.hide)` annotation? What's the big secret? This annotation prevents even the documentation from being visible.

Complex should "feel" like a built-in type (such as "Real", "Integer"). 
There is also no top-level "Real" definition in the package browser and therefore there should not be a Complex definition visible to the user.
This is also the reason why Complex only contains the basic mathematical operation and all other operations on Complex numbers are in Modelica.ComplexMath

modelica-trac-importer, 2017-01-15:
**Comment by stefanv on 6 Nov 2015 14:02 UTC**
I think choosing to hide it in this way is tool-specific. In MapleSim for instance, it already doesn't show up in the component browser because it doesn't contain any components. However, if a user _does_ want to look inside it, they currently can't, because of the arbitrary (in my opinion) decision to hide it.

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 6 Nov 2015 22:01 UTC**
I agree and it is actually ironic that it is now tool-specific since the original idea was to clean out `__Dymola_Protection(hideFromBrowser=true)` from the MSL. Question is what better alternative is there? Should we leave this decision to the tool (like you suggest) or to the library developer (like it now is intended though I'd argue that not being able to look at the content was not the intention). Btw the ticket were this was originally discussed is #585.

modelica-trac-importer, 2017-01-15:
**Comment by ahaumer on 7 Nov 2015 14:56 UTC**
I think we should discuss whether to hide Complex from the user or not ...

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 9 Nov 2015 15:20 UTC**
Replying to [comment:5 ahaumer]:
> I think we should discuss whether to hide Complex from the user or not ...

Well, I think there are two different questions:
* Should it be possible to see the contents of Complex when needed. I believe the answer is clearly 'Yes'.
* Should 'Complex' as default be shown in the package browser? That is less clear, but having it as visible as 'Modelica' seems a bit too much.
(and similarly for ModelicaServices).

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 8 Jan 2016 19:07 UTC**
Ticket retargeted after milestone closed

HansOlsson, 2018-01-15:
To clarify:

Well, I think there are two different questions:
- Should it be possible to see the contents of Complex when needed. I believe the answer is clearly 'Yes'.
- Should 'Complex' as default be shown in the package browser? That is less clear, but having it as visible as 'Modelica' seems a bit too much. (and similarly for ModelicaServices - but it is anyway tool-specific).

I believe the second question can be left to tool-vendors without the Complex-library having  `Protection(access=Access.hide)`.

tobolar, 2018-01-15:
In my opinion the same applies also for ModelicaServices.

casella, 2018-01-22:
@otter wrote: "Complex should "feel" like a built-in type (such as "Real", "Integer").".

In principle the idea is good, but unfortunately Complex is not a built-in type, so if I look for it in the specification, Section 4, there is no such thing as complex. I don't think having Complex work like magic is a good idea either. Where can I look for the documentation of this almost-built-in type, if not in its documentation layer?

So, I agree with @HansOlsson that Complex should not be hidden. 

I also agree with @tobolar that  ModelicaServices should not be hidden as well: many classes in the MSL refer to ModelicaServices, and not seeing it anywere sounds very odd to me.

christiankral, 2018-01-24:
I agree with @casella that Complex shall NOT be hidden. Since complex quantities are unfortunately not built in, the user has to see and understand how the implementation is done. 

If the complex quantities were built in, the functionality were described in the Modelica specification and we would not have to deal with the conversion of variables from rectangular to polar form and vice versa.

beutlich, 2018-04-30:
> In my opinion the same applies also to ModelicaServices.

There is one major difference between packages Complex and ModelicaServices, that ModelicaServices depends on Modelica whereas Complex does not. In consequence, it is no longer possible to reload package Modelica in the Library Browser of SimulationX since ModelicaServices depends on it. (It was possible when ModelicaServices was hidden in Library Browser though). I therefore propose to add the protection annotation back to ModelicaServices.

tobolar, 2018-04-30:
On the other side it is not possible to reload ModelicaServices (for whatever reasons) when it is protected. Or at least in Dymola "attempted redefinition of ModelicaServices was ignored".

beutlich, 2018-04-30:
There should be no need for users to reload ModelicaServices as it is provided by the tool vendor.

sjoelund, 2018-04-30:
It seems that SimulationX provides their own ModelicaServices and could patch the protected annotation if necessary. It's very useful to show the package since MSL extends from it.

beutlich, 2018-04-30:
Agreed.

commit 76c984c9 (2018-04-23): Merge pull request #2504 from dietmarw/complex

Fix #1822 by removing the Protection annotation and bump version
commit ca8476d7 (2018-04-23): Fixes #1822 by removing the Protection annotation
closed 2018-04-30 by beutlich