modelica/ModelicaStandardLibrary #1839: CombiTable1D extrapolates wrong when setting ConstantSegments

modelica-trac-importer, 2017-01-15:
**Reported by markus&#x2e;andres on 18 Nov 2015 17:13 UTC**
When simulating the code below in Dymola with default settings for solver etc. the "tableWrong" seems to extrapolate so negative values.
Adding an artificial point corrects the behavior although this should not be the case with this type of extrapolation in my opinion.

```
model ModelicaBug

  Modelica.Blocks.Tables.CombiTable1D tableWrong(smoothness=Modelica.Blocks.Types.Smoothness.ConstantSegments, table=[0,0; 1000, 0.02; 6000,0]) annotation (Placement(transformation(extent={{-10,20},{10,40}})));
  Modelica.Blocks.Sources.Ramp ramp(height=10000, duration=1000)  annotation (Placement(transformation(extent={{-60,-10},{-40,10}})));
  Modelica.Blocks.Tables.CombiTable1D tableGood(smoothness=Modelica.Blocks.Types.Smoothness.ConstantSegments, table=[0,0; 1000, 0.02; 6000,0; Modelica.Constants.inf,0]) annotation (Placement(transformation(extent={{-10,-40},{10,-20}})));
equation
  connect(ramp.y, tableWrong.u[1]) annotation (Line(points={{-39,0},{-26,0},{-26,30},{-12,30}}, color={0,0,127}));
  connect(tableGood.u[1], ramp.y) annotation (Line(points={{-12,-30},{-26,-30},{-26,0},{-39,0}}, color={0,0,127}));
  annotation (
    uses(Modelica(version="3.2.1")),
    experiment(StopTime=1000),
    __Dymola_experimentSetupOutput);
end ModelicaBug;
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1839

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 18 Nov 2015 19:36 UTC**
Though being unexpected probably, it actually is by design that CombiTable1D, CombiTable1Ds and CombiTable2D always extrapolate linearly no matter how the interpolation is set.

Block ModelicaTest.Tables.CombiTable2D.Test19.combiTable1D is a test example for it. In #1313 I also raised this topic (under wrong assumptions) but nobody replied.

Not sure what to do since linear extrapolation for constant interpolation rather feels non-naturally.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 18 Nov 2015 19:43 UTC**
And it's properly documented by the way:
```
If the input signal u is outside of the defined interval,
i.e., u > table[size(table,1),1] or u < table[1,1], the
corresponding value is also determined by linear interpolation
through the last or first two points of the table. 
```

modelica-trac-importer, 2017-01-15:
**Comment by markus&#x2e;andres on 23 Nov 2015 10:32 UTC**
Replying to [comment:2 beutlich]:
> And it's properly documented by the way:
> ```
> If the input signal u is outside of the defined interval,
> i.e., u > table[size(table,1),1] or u < table[1,1], the
> corresponding value is also determined by linear interpolation
> through the last or first two points of the table. 
> ```

Hm off topic but still: I'm having trouble to comment on the topic as trac tell's me I'm writing spam. I'm told to answer a question below that is not showing up. 

Rearging the topic: True that it is documented well, but I have to support your view that it "rather feels non-naturally". Would it be possible to add an option to have linear or constant extrapolation?

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 23 Nov 2015 15:03 UTC**
Yes, the spam trap of this trac actually is annoying.

Just my concerns:
 * If we change it we also should do it for 2d table.
 * We might even change it w/o adding a new option but claiming that extrapolation should (better) comply with interpolation.

modelica-trac-importer, 2017-01-15:
**Comment by markus&#x2e;andres on 23 Nov 2015 15:53 UTC**
Replying to [comment:5 beutlich]:

>  \* We might even change it w/o adding a new option but claiming that extrapolation should (better) comply with interpolation.

I would be totally happy with that solution as well.

> Yes, the spam trap of this trac actually is annoying.

This is my fourth try to answer, basically just adding stuff to trick that filter ;)
But it seems to not like my messages at all. Though pressing Reply instead of add comment seems to help...

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 23 Nov 2015 21:26 UTC**
I started a poll about the potential change alternatives to gain more feedback from the MSL shapers and users. Feel free to contribute (by the end of next week).

http://doodle.com/poll/4yauvrg2x92eymtf

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 5 Jan 2016 10:40 UTC**
This requires new API functions fro ModelicaStandardTables. I will wait for this API change to also incorporate the getInstanceName.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 9 Aug 2016 11:40 UTC**
Voting results: 7 votes for "Add a new extrapolation option (like CombiTimeTable has) that offers three alternatives for extrapolation: Linear, Constant, No."

beutlich, 2017-06-20:
@markusandres3DS Extrapolation for CombiTable1D was added by fb449bb2fb942887f349adb38938a6dc86aa4436 (See also #2120). Extrapolation for CombiTable2D is left TODO, however the API functions already were prepared by acd6ac4a273d661426db578e3a0f8449e5805e12.

beutlich, 2017-08-25:
@markusandres3DS Extrapolation for CombiTable2D was added by 9e15bbf117d39d23814b903ffbc15a491ce17719.

closed 2017-08-25 by beutlich