modelica/ModelicaStandardLibrary #1843: Make Modelica.Mechanics.MultiBody.Visualizers.Ground usable for x- and y direction

modelica-trac-importer, 2017-01-15:
**Reported by johanr on 24 Nov 2015 21:48 UTC**
The component `Modelica.Mechanics.MultiBody.Visualizers.Ground` only supports a plane in z=0. 
There should be a parameter with the values `z=0`, `y=0` and `x=0` to adjust which plane is used.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1843

tobolar, 2018-01-16:
I don't fully understand the idea of the three values.
We can define the ground either by n_x and n_y vectors or by its normal vector, such as 

    parameter Modelica.Mechanics.MultiBody.Types.Axis n = {0,0,1};

Moreover, the visualization of the ground plane by a cube is odd and I would suggest to use Plane surface (added in #2424) instead.

beutlich, 2018-01-17:
The comment of Modelica.Mechanics.MultiBody.Visualizers.Ground stiil reads "Visualizing the ground (box in z=0)". Also the icon shows the z axis.
[image]

beutlich, 2018-01-18:
* Should Ground.height be set disabled?
* And why not setting it to zero?
* Also, I do not like the screaming OBSOLETE. Just keep the comment as was.
* And there should be an issue to remove Ground.height for the next non-backward compatible MSL.

beutlich, 2018-01-18:
That new normal Integer parameter is kind of odd. Why not using a parameter of type Modelica.Mechanics.MultiBody.Types.Axis to specify the normal axis?

tobolar, 2018-01-18:
I agree with your points and improved it in 9c5cb4998a49ee92e6af3f14aa154dbb579ff5ae.

>And there should be an issue to remove Ground.height for the next non-backward compatible MSL.

Is this done with your comment?

>That new normal Integer parameter is kind of odd. Why not using a parameter of type Modelica.Mechanics.MultiBody.Types.Axis to specify the normal axis?

I suppose the intention of the Ground model is to simply visualize a xy, xz or yz plane. Once the normal should be defined by a vector, then the model would became more complex since we need one more vector to define the plane. Was this the original idea of the model?

HansOlsson, 2018-01-18:
> I suppose the intention of the Ground model is to simply visualize a xy, xz or yz plane. Once the normal should be defined by a vector, then the model would became more complex since we need one more vector to define the plane. Was this the original idea of the model?

On the other hand visualizing an inclined plane instead of just a xy, xz, or yz plane seems like a straightforward extension. Obviously it is a trade-off between having everything and make it easy to understand for users.
I also realized that the default for the ground-normal should ideally be 'g' from the world-object and not the z-plane, since ground is as default orthogonal to gravity.

tobolar, 2018-01-19:
Actually, there are two possibilities:
1. a simple ground with center placed at the origin of the world and oriented in xy, xz, or yz axes,
2. a generic ground plane with arbitrary orientation and center point position.

Even if there is a relation to the World, the Ground is still a particular visualizer and that's why I see it as a simple model. If more advanced visualization is desired, there can still be used `Modelica.Mechanics.MultiBody.Visualizers.Plane` instead.

Another option is to make the groud plane a part of the World, see #2432. 

> I also realized that the default for the ground-normal should ideally be 'g' from the world-object and not the z-plane, since ground is as default orthogonal to gravity.

Well this is just for compatibility reason to the previous implementation.

MartinOtter, 2018-01-19:
Since ground should be orthognol to the gravity vector, it seems simplest for the user to have the ground in the center of world in a plane that is orthogonal to the gravity vector (inquired via outer World word). Unfortunately, this would be a non-backwards compatible change.
On the other hand, "ground" is just a box and a user can easily use a FixexShape box and place it arbitrarily.
My conclusion: Either close this ticket without changes to MSL, or introduce a new Ground object that is orthogonal to the gravity vector (in order to be backwards compatible).

beutlich, 2018-01-22:
>>And there should be an issue to remove Ground.height for the next non-backward compatible MSL.

>Is this done with your comment?

No, there should be an extra issue targeting [MSL_next-MAJOR-version](https://github.com/modelica/Modelica/milestone/4). See #1202 for a similar issue.

tobolar, 2018-01-22:
@MartinOtter  I agree and am going to make the ground plane a part of the World model, see #2432.

@beutlich OK

HansOlsson, 2018-03-06:
> Can you please motivate your decision? It is hard to keep up with all those back and forth changes.

I agree that it is hard to follow, I believe it would help to just have list of the different variants that we have considered.

tobolar, 2018-03-06:
I suppose the idea of the Ground visualizer was to have more or less strong link to the World component. As the ground plane is defined directly in the World by  66345025f76bdfbb06ac7cd33e53707a25402718, and there is aditionally a new component `Modelica.Mechanics.MultiBody.Visualizers.Plane`, the Ground visualizer becomes obsolete to my opinion.
With other words, when someone wants to visualize the ground plane, he/she can do it in the World directly. If more flexibility is needed then the Plane visualizer can be used. So there is no need to have a particular ground visualizer in the multi-body.

I don't stick on reverting the changes. More important to my opinion is to remove the component in the next major version of the library.

commit e13e532e (2018-01-18): Close #1843: MultiBody Ground is visualized as a plane instead of a box; the plane normal can be changed
closed 2018-04-16 by tobolar