modelica/ModelicaStandardLibrary #1859: Proposal of adding "Chemical" into Modelica Standard Library

modelica-trac-importer, 2017-01-15:
**Reported by marek on 16 Dec 2015 10:09 UTC**
We want to propose a Chemical library (https://github.com/MarekMatejak/Chemical) to be a part of a Standard Modelica Library. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1859

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 16 Dec 2015 11:09 UTC**
There are some who feel the MSL getting too bloated and even harder to maintain. Adding yet another domain library into the MSL is not helping this. So I would suggest the Chemical library is kept separate and encapsulated so that it can be easily integrated in a future modularised MSL.

modelica-trac-importer, 2017-01-15:
**Comment by Marek Matejak on 29 Feb 2016 13:05 UTC**
We found very useful to have at least only standardized connectors for chemical domain in Modelica such as Chemical.Interfaces.SubstancePort and Chemical.Interfaces.SolutionPort, because only the user defined classes using these connectors will be compatible together. However, there are already the base components of each domain in MSL, so we do not see the reason why the chemical domain cannot be part of the Modelica Standard Library. Please note, that there are not sufficient support for graphical modeling of electro-chemical processes even using the next MSL 3.2.2.

Of course our library must be evaluated for inclusion in the Modelica Standard Library (MSL) according to the current processes of the Modelica Association Project "Modelica Libraries". If it will be found appropriate for inclusion in the MSL  (and fulfills all quality requirements), it needs to be accepted by the MAP "Modelica Libraries" group. If accepted, it will be listed on the MA Library Web page under "Libraries developed by the MA" and it will be kept as a separate library for "some time" (time period needs to be discussed). It must be in the "right design" and is "good" enough. It should be also reviewed by at least two experts in physical chemistry.

modelica-trac-importer, 2017-01-15:
**Comment by choeger on 15 Mar 2016 13:48 UTC**
This seems (I am definitely not an expert) to be a great library. However, I support Dietmar's point of view: There is no need to bloat the MSL further. And "other domains are present, too" is not an argument. If we want to stop the growth of the MSL, we should start right here and in turn make a point of refactoring all other domains as well. 

There is no reason that the MA can only endorse one library. Instead let us have one standard library and lots of different domain libraries!

AHaumer, 2018-01-04:
Why should we stop the MSL grow? 
If we remove all domains the MSL is rather small. 
Splitting MSL in domain libraries is a great idea. 
Unfortunately up to now I didn't see a mechanism that handles dependencies of these libraries when the user decides to update MSL.

mtiller, 2018-01-05:
@AHaumer it isn’t that we don’t want the MSL to grow, it is that we want to establish a more scalable approach to grow.  And that scalability doesn’t just apply to the MSL but to the whole eco-system.  Ultimately, this all helps growth.

dietmarw, 2020-01-24:
Having worked actively with the Chemical library I can say that it is far from finished and really not in the shape to be included in the MSL (principle discussion aside). So I'm closing this. Should this become relevant again and MAP-LIB has the resources a new ticket should be opened.

closed 2020-01-24 by dietmarw