modelica/ModelicaStandardLibrary #1860: "class" still a valid Modelica class type to use?

modelica-trac-importer, 2017-01-15:
**Reported by dietmarw on 16 Dec 2015 12:05 UTC**
I remember there were earlier discussion(s) on the Modelica class type `class` that should be discouraged to be used. Unfortunately I could not find a specific ticket on this. Now I was wondering should one in future try to avoid using `class` in all circumstances or are there still valid use-cases. 

If it should be avoided then the next question would be what _class_ the documentation class with `DocumentationClass=true` annotation should be. `model`? 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1860

beutlich, 2018-03-21:
@HansOlsson 
Library group: This looks like a specification issue if class = model (as in Dymola) or if class is an abstract term for all specialized classes only. Currently tools behave differently here, which means some clarification is required.

Proposal: Introduce new specialied class for documentation-only classes.

christiankral, 2018-03-27:
In a record it is not allowed to use implicit equations in an ```initial equation``` section. This is often a drawback, especially, when complex calculations shall be applied. 

Alternatively a ```class``` can be used as the restriction of not having an ```initial equation``` section does not apply:

```modelica
package Testing
  class Mo
    parameter Real x(fixed=false);
  initial equation 
    x=cos(x);
  end Mo;

  model TestMo
    parameter Testing.Mo mo;
  end TestMo;
end Testing;
```

Is there an alternative way of implementing implicit equations as parameter (class) record?

beutlich, 2018-03-27:
Section 4.4.2.2 of MLS 3.4 reads:
> Type prefixes (that is ,  flow ,  stream ,  discrete ,  parameter ,  constant ,  input ,  output ) shall only be
applied for type, record and connector components – see also record specialized class, Section 4.6.

It does not mention class there as allowed type. To me it seems we should legalize the use case if @christiankral.

christiankral, 2018-03-27:
Hmm... I thought my example this is legal, but it is obviously  not. 

However, the example ```TestMo``` compiles and simulates both in Dymola and OpenModelica without a warning.

To make the example more concrete, I renamed class ```Mo``` to ```Data``` to better represent the intended case. What I would like to have is basically a record which allows implicit equations. So the intention is (using ```class``` as a synonym for a record which allows implicit equations):

```modelica
package Intention
  class Data
    parameter Real z=1;
    parameter Real x(start=0,fixed=false);
  initial equation 
    x=cos(z*x);
  end Data;

  class Data2
    extends Data(z=2);
  end Data2;

  model Component
    parameter Data data;
  end Component;

  model TestMo1
    parameter Data2 data;
    Component component(data=data);
  end TestMo1;

  model TestMo2
    parameter Data2 data;
    Component component;
  initial equation 
    component.data=data;
  end TestMo2;
end Intention;
```

Without the parameter prefix I cannot modify the parameter ```data``` of a component. Instead I have to use an initial equation. When propagating only one parameter this seems not be a lot of effort, but for the application I have in mind, it were a lot of parameter which had to be propagated. 

```TestMo1``` is what I would like to have, but this seems not to be legal in Modelica. ```TestMo2``` is some kind of a compromise; at least I can propagate all parameters within one line of the ```initial equation```.

If I remove the ```parameter``` prefix from the implicit equation of ```Data``` as proposed by @beutlich I can do the following:

```modelica
package Testing
  model Data
    parameter Real z=1;
    Real x;
  equation 
    x=cos(z*x);
  end Data;

  class Data2
    extends Testing.Data(z=2);
  end Data2;

  model Component
    parameter Real x(fixed=false);
  end Component;

  model Example
    Data2 mo;
    Component component;
  initial equation 
    component.x=mo.x;
  end Example;
end Testing;
``` 

Since ```Component``` shall use a parameter, then, in the initial equation I have to propagate each variable separately. For only one parameter this seems not be a lot of effort, but for the application I have in mind, it were a lot of parameter which had to be propagated.

beutlich, 2019-01-24:
We have two use cases here.
1. Documentation only class
1. Parameter class with initial equation

@HansOlsson Can you please make specification issues out of this. Thanks.

HansOlsson, 2019-01-24:
I have opened the specification ticket, but there are parts that are not clear.

> Without the parameter prefix I cannot modify the parameter data of a component. Instead I have to use an initial equation. When propagating only one parameter this seems not be a lot of effort, but for the application I have in mind, it were a lot of parameter which had to be propagated.

There are a number of issues hidden here. When you have a data-class with equations each such component will have initial equations as well. Thus for 
```
package Intention
  class Data
    parameter Real z=1;
    parameter Real x(start=0,fixed=false);
  initial equation 
    x=cos(z*x);
  end Data;

  model Component
    parameter Data data;
  end Component;

  model Test
    parameter Data data(z=2);
    Component component(data=data);
  end Test;
end Intention;
```
We will for `Intention.Test` have the following:
```
  data.z=2;
  component.data.x=data.x;
  component.data.z=data.z;
initial equation
  data.x=cos(data.z*data.x);
  component.data.x=cos(component.data.z*component.data.z);
```
That is 5 equations for 4 parameters, which doesn't work well. 

Relatedly we don't want to be presented with a parameter-dialog for "Data", or be led to use a record-constructor for it - since we are only supposed to set z.

As I see it there are two possibilities:

- Have some special semantics for this. We need to specify this further.
- Handle it within the current language, e.g. by construct a record and a model for such cases:
```
package Intention
  record Data
    parameter Real z=1;
  end Data;

  model ComputedData
   parameter Data data;
   parameter Real x(fixed=false, start=0);
  initial equation
    x=cos(data.z*x);
  end ComputedData;

  model Component
   parameter Data data;
   ComputedData computed(data=data);
  end Component;

  model Test
    parameter Data data(z=2);
    Component component(data=data);
  end Test;
end Intention;
```
I see a slight issue in the code above that ComputedData distinguishes between the given data and the computed ones (it is also possible to avoid that - but that complicates Component). I understand that makes it less user-friendly, so it seems that we might need something related to https://github.com/modelica/ModelicaSpecification/issues/1953 

(I also see that there are more parameters overall but they are just aliases so tool can handle that - similarly as aliases for connector-variables.)