modelica/ModelicaStandardLibrary #1861: Noise: Improve automatic local seed function

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 16 Dec 2015 21:03 UTC**
This simply is a reminder for the following note in Noise.mo
```
It is planned to change the automatic local seed function
in a future version of package Modelica, once Modelica Language 3.3
language elements can be used ...
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1861

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 13 Oct 2016 20:46 UTC**
Can this be closed as obsolete after resolving #1957 for MSL 3.2.2?

beutlich, 2018-11-12:
@MartinOtter Is this still planned?

MartinOtter, 2019-01-11:
I would suggest to first have tools with MSL 3.2.3 support and only then improve this

dietmarw, 2019-01-11:
@MartinOtter I guess you mean MLS 3.3 support, correct?

MartinOtter, 2019-01-11:
> 
> 
> @MartinOtter I guess you mean MLS 3.3 support, correct?

Yes (sorry).

PR #3224 (2019-11-13): Add automaticLocalSeed in Math.Random and utilize in noise blocks
There was some in and out development going on in #1957 when the noise blocks and functions were introduced in MSL v3.2.2 with the automaticLocalSeed left to do. This pull request fixes #1861.

@akloeckner @dzimmer @fvanderlinden FYI
closed 2019-11-13 by beutlich