modelica/ModelicaStandardLibrary #1865: Consistent use of terms multi-phase and single-phase

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 16 Dec 2015 23:07 UTC**
The terms multi phase / single phase and multiphase / singlephase are used within the Electrical library. It would be more standard-like if they would be used consistently. Also multi-phase / single-phase are possible. Some native speakers may advise here.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1865

modelica-trac-importer, 2017-01-15:
**Comment by ahaumer on 18 Dec 2015 12:13 UTC**
You're right. 
Since this is no big issue, I'm waiting for native speakers' advise (shifting the mielstone).

ceraolo, 2017-01-28:
I think that more than native speakers what counts is IEC's opinion.
On the International Electrotechnical vocabulary we find single-phase :
http://dom2.iec.ch/iev/iev.nsf/SearchView?SearchView&Query=field+SearchFields+contains+single-phase+and+field+Language=en&SearchOrder=4&SearchMax=0

beutlich, 2017-08-05:
So: single-phase and polyphase are to be used, right?

ceraolo, 2017-08-07:
I agree.
IEV is the highest source of information for electricity terms.
They use single-phase and polyphase.

beutlich, 2017-09-19:
I closed #2290 due to plenty of merge conflicts. Feel free to reopen in case you want to have consistent terms one day.

closed 2017-09-19 by beutlich

christiankral, 2017-09-20:
I discussed this issue with @ahaumer and double checked with my English text books on electric machines and systems. So on my opinion the term polyphase seems to be the right one to be used. After implementing the terms in the MSL correctly, we will have to consider renaming the sub library Electrical.MultiPhase to Electrical.PolyPhase in the next major release.  

@beutlich Sorry for corrupting your proposal. I was not not aware of that when I worked on #2276.