modelica/ModelicaStandardLibrary #1866: Fix standalone HTML format of Modelica License 2

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 17 Dec 2015 21:01 UTC**
The provided [html format of the Modelica License 2](https://modelica.org/licenses/ModelicaLicense2.html/at_download/file) misses some closing html tags. I run HTML Tidy with options `--doctype "loose" --logical-emphasis y --indent "auto" --indent-spaces "4" --wrap "80"` to fix these (and some other compliance issues). Note that with option `--logical-emphasis` <i> is replaced by <em> and <b> by <strong> as proposed by the MSL naming conventions and reported in #1747.

This updated license should be
 * provided at the official download at https://modelica.org/licenses/ModelicaLicense2.html/view and https://modelica.org/licenses/ModelicaLicense2.html/at_download/file
 * deployed by package.mo of MSL 3.2.2 and ModelicaServices
 * deployed by all other Modelica_\* packages under Modelica License 2

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1866

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 18 Dec 2015 14:40 UTC**
Could you run tidy once again with out touching the indent? That makes the diff clearer as the changes in the HTML need also be adapted in the mo files.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 18 Dec 2015 15:38 UTC**
Replying to [comment:2 dietmarw]:
> Could you run tidy once again with out touching the indent? That makes the diff clearer as the changes in the HTML need also be adapted in the mo files.

HTML Tidy has no option to keep the existing indents. Since the current license file has mixed-style indents you will always get modified indents in the tidied license file, either no indents at all or consistent indents.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 18 Dec 2015 22:13 UTC**
Please read also http://www.w3.org/People/Raggett/tidy:
"You can choose which style you want Tidy to use when it generates the cleaned up markup: for instance whether you like elements to indent their contents or not. Several people have asked if Tidy could preserve the original layout. I am sorry to say that this would be very hard to support due to the way Tidy is implemented. Tidy starts by building a clean parse tree from the source file. The parse tree doesn't contain any information about the original layout. Tidy then pretty prints the parse tree using the current layout options. Trying to preserve the original layout would interact badly with the repair operations needed to build a clean parse tree and considerably complicate the code."

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 21 Dec 2015 22:34 UTC**
OK I replaced the license file in the source of the ModelicaDesign repo in r8848 (also replaced on https://modelica.org/licenses). But it would be nice if you could run tidy once again on the html from the applied license text in the MSL.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 22 Dec 2015 08:28 UTC**
Dietmar, I found some issues in the MoLic2 (Modelica Format):
 * Pair of \" used where &ldquo; and &rdquo; should be used instead
 * Missing question mark and blank after "Can I copy a type class or part of a model, block, function, record, connector class, out of a Modelica package (under Modelica License 2) and include it modified or unmodified in a Modelica package under a commercial/proprietary license?"
 * One single &rsquo; used where it should be a &apos;
 * Non-matching pair of &ldquo; and &rdquo; used
 * Missing non-breaking space before number 2, e.g. Modelica License 2 should be Modelica License&nbsp;2
 * Add question mark (instead of colon) to "Can I sell a manual that was basically derived by extracting information automatically from a Modelica package under Modelica License 2 (e.g., a “reference guide” of the Modelica Standard Library):"

I attached the original file https://modelica.org/licenses/ModelicaLicense2.mo/at_download/file, the fixed file and the fixed+tidied file of the Modelica format.

I guess I need to consider the HTML Format and the ones from the MSL again.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 22 Dec 2015 10:52 UTC**
Fixed for MSL trunk by 91434cdab57279f90833e49da1463bbf8746dd7b.

Also fixed for Modelica_DeviceDrivers by [84ea6d1](https://github.com/modelica/Modelica_DeviceDrivers/commit/84ea6d189d3c0870544f2510903963fb241ecb56).

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 22 Dec 2015 16:34 UTC**
So I guess the fixed and tidied version should be placed in the ModelicaDesign repo? What about the html does it have the same problems you describe above?

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 22 Dec 2015 18:20 UTC**
Yes, it has. Will provide a fix for HTML, too. But, PDF, Word and OpenDoc need to be updated, too.

Question: There is <li></li> without exlicit <p></p> in "1. Definitions.". But <p></p> is then used within the list in "2. Grant of Copyright License.". This looks not really consistently.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Feb 2016 08:35 UTC**
Replaced the fancy quotes “ and ” by a standard one " by 5ecf888466f1c04e428a270fb151b3ac9197a733.

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 5 Feb 2016 17:17 UTC**
New version added to the Modelica Design repo in r9042. Also updated https://modelica.org/licenses/

Mind I did not touch the odt, doc, or PDF files.

modelica-trac-importer, 2017-01-15:
**Comment by anonymous on 21 Nov 2016 07:44 UTC**
TODO in HTML license: Make list formatting consistent: There is <li></li> without exlicit <p></p> in "1. Definitions.". But <p></p> is then used within the list in "2. Grant of Copyright License."

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 20 Dec 2016 18:29 UTC**

beutlich, 2017-02-13:
> Question: There is  `<li></li>`  without exlicit  `<p></p>`  in "1. Definitions.". But  `<p></p>`  is then used within the list in "2. Grant of Copyright License.". This looks not really consistently.

Resolved in master and maint/3.2.2 branches.

closed 2017-02-13 by beutlich