modelica/ModelicaStandardLibrary #1869: enhancement of Modelica.Utilities.Examples.expression

modelica-trac-importer, 2017-01-15:
**Reported by ahaumer on 19 Dec 2015 22:30 UTC**
Modelica.Utilities.Examples.expression is a usefull function that could easily get enhanced:

implement power:
function term
...
  result := if opString == "\*" then result \* result2 else _if opString == "/" then result / result2 else result ^ result2_;
...

implement additional functions asin(), acos, atan, exp, log:
function primary
...
          result := sqrt(result);
_        elseif functionName == "asin" then
          result := Math.asin(result);
        elseif functionName == "acos" then
          result := Math.acos(result);
        elseif functionName == "atan" then
          result := Math.atan(result);
        elseif functionName == "exp" then
          result := Math.exp(result);
        elseif functionName == "log" then
          if result <= 0.0 then
            Strings.syntaxError(string, startIndex, "Argument of call \"log(" + String(result) + ")\" is negative.\n" + message);
          end if;
          result := log(result);_
...

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1869

beutlich, 2017-10-14:
@AHaumer #2365 addresses asin(), acos(), atan(), exp(), log() in function primary. But adding `^` is non-trivial since it requires structural changes of the grammar. Your proposed change to function term is insufficient.

beutlich, 2018-01-09:
61661e2376da56dca221b5d4f397c0ca9469029c added functions asin(), acos(), atan(), exp(), log(). But the power function was not added.

closed 2018-01-09 by beutlich