modelica/ModelicaStandardLibrary #1871: Why start value for Modelica.Electrical.Analog.Interfaces.ConditionalHeatPort.heatPort.T?

modelica-trac-importer, 2017-01-15:
**Reported by ahaumer on 21 Dec 2015 18:28 UTC**
Modelica.Electrical.Analog.Interfaces.ConditionalHeatPort is implemented slightly different from Modelica.Thermal.HeatTransfer.Interfaces.PartialElementaryConditionalHeatPort:
parameter Modelica.SIunits.Temperature T;
Modelica.SIunits.Temperature T_heatPort;
Modelica.Thermal.HeatTransfer.Interfaces.HeatPort heatPort(T(start=T)...) 
Why this start value? Different from HeatTransfer!
if useHeatPort=false
   T_heatPort=(parameter) T
   heatPort not existing
if useHeatPort=false
   T_heatPort=heatPort.T
   parameter T unused

If the user sets useHeatPort=true, and prescribes a temperature heatPort.T, using a different start value = parameter T this start value is in conflict with the prescribed temperature. 
In pedantic mode, Dymola 2016 generates an error during translation; Dymola 2016FD01 generates a warning during translation but simulates.

In my opinion we can safely remove the start value heatPort(T(start=T)), and we should do that.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1871

beutlich, 2017-04-19:
Resolved in master.

closed 2017-04-19 by beutlich

beutlich, 2017-08-28:
Also affects Modelica.Magnetic.FluxTubes.Interfaces.ConditionalHeatPort.