modelica/ModelicaStandardLibrary #1873: Desired mechanical components

modelica-trac-importer, 2017-01-15:
**Reported by ahaumer on 26 Dec 2015 10:02 UTC**
Two mechanical components are missing:
* a model of drive pulley
* a simpler model of a gearbox with constant efficiency
The Mechanics.Rotational.Components.LossyGear is not that easy to paraeterize and use for beginners; this is easier with the proposed gearbox with constant efficiency. 
Furthermore, adrive pulley can be modelled using Mechanics.Rotational.Components.IdealRollingWheel; though connections and parameterization of cabin and counterweight of an elevator is not that easy. 
The drive pulley can be modelled as an IdealDrivePulley or with constant efficiency (like the gearbox mentioned above). 
For cabin and counterweight, the mass is modelled together with the gravitational force. 
See the enclosed library which also contains an example of usage. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1873

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 15 Apr 2016 11:12 UTC**
See also #1778 for another missing mechanical component.

tobolar, 2018-01-15:
Another missing component is in #506.

tobolar, 2020-01-10:
No chance to do this by the feature freeze of MSL v4.0.0 development.

MartinOtter, 2020-01-11:
> 
> 
> **Reported by ahaumer on 26 Dec 2015 10:02 UTC**
> Two mechanical components are missing:
> 
>     * a model of drive pulley
> 
>     * a simpler model of a gearbox with constant efficiency
> 
> 
> The Mechanics.Rotational.Components.LossyGear is not that easy to parameterize and use for beginners; this is easier with the proposed gearbox with constant efficiency.
> Furthermore, adrive pulley can be modelled using Mechanics.Rotational.Components.IdealRollingWheel; though connections and parameterization of cabin and counterweight of an elevator is not that easy.
> The drive pulley can be modelled as an IdealDrivePulley or with constant efficiency (like the gearbox mentioned above).
> For cabin and counterweight, the mass is modelled together with the gravitational force.
> See the enclosed library which also contains an example of usage.
> 
> [Elevator.mo](https://github.com/modelica/Modelica/files/744240/Elevator.mo.txt)
> 
> Migrated-From: https://trac.modelica.org/Modelica/ticket/1873

I agree that we should provide a simpler model for a Gearbox with efficiency. The problem with your proposal is that you use a constant efficiency: When inspecting a gearbox catalogue, then there are diagrams of the efficiency over speed, and for low speed the (catalogue) efficiency goes to zero, so a typical efficiency from a gearbox catalogue is far from constant. Can we phone us next week to discuss this. One option could be that the user provides the efficiency at two speeds (say nominal speed and the lowest speed shown in the catalogue). From this information, an instance of LossyGear is constructed.