modelica/ModelicaStandardLibrary #1887: Extracted HTML help for Modelica.SIunits is nearly useless

modelica-trac-importer, 2017-01-15:
**Reported by stefanv on 22 Jan 2016 18:53 UTC**
Every entry in the file `help/Modelica_SIunits.html` has a table of the form:

|**Name** | **Description**|
|-|-|
|quantity|
|unit|

In every case, the second column is empty, which is not surprising, since there's no way to add a string comment to the already existing `Real.quantity` field where the description presumably comes from.

The second column should be **Value**, not **Description**.

It might be useful to also include the min and max values in this table.

For example:

|**Name** | **Value**|
|-|-|
|quantity | "Mass"|
|unit | "kg"|
|min | 0|
|max | Modelica.Constants.inf|

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1887

dietmarw, 2017-01-16:
I'd say this is a tool issue of the HTML generator. So currently either Dymola or OpenModelica

HansOlsson, 2017-02-07:
The help generated by Dymola and 3D Experience Platform will replace the Description-column with a value-column. However, as far as I can tell the generated help-directory is not include on github.

beutlich, 2017-02-07:
> However, as far as I can tell the generated help-directory is not include on github.

It is. See branch gh-pages, i.e., https://github.com/modelica/Modelica/tree/gh-pages/help which is used to build http://doc.modelica.org.

HansOlsson, 2017-02-07:
> > However, as far as I can tell the generated help-directory is not include on github.
> 
> It is. See branch gh-pages, i.e., https://github.com/modelica/Modelica/tree/gh-pages/help which is used to build http://doc.modelica.org.

Ah, I see. It wasn't on the main page

dietmarw, 2017-02-07:
@HansOlsson The help files are generated files, so no point putting them into the development version of the MSL. They are normally only included in the released zip-balls. Though not included in the latest ones since Dymola keeps putting them in the wrong hierarchy as you know. They are however put into the gh-pages branch to allow online browsing.

beutlich, 2017-02-07:
> Ah, I see. It wasn't on the main page

The MSL documentation of the generated Dymola and OM help is linked from the [README](https://github.com/modelica/Modelica/blob/master/README.md) on the main page.

beutlich, 2017-02-07:
@HansOlsson If you have a newer help of MSL 3.2.2 coming with a Value description of the SI units then you are welcome to push against gh-pages branch to update the online documentation. Thank you.

beutlich, 2017-02-14:
@HansOlsson Do you have any plans to update the Dymola generated help available at http://doc.modelica.org.?

HansOlsson, 2017-02-14:
> @HansOlsson Do you have any plans to update the Dymola generated help available at http://doc.modelica.org.?

Yes I was planning on updating that branch - which I understand is used for that.
But some other things were more urgent.

beutlich, 2017-04-18:
@HansOlsson @svorkoetter Finally resolved in branch gh-pages, though my comment on short class definitions was not considered.

closed 2017-04-18 by beutlich