modelica/ModelicaStandardLibrary #1891: Add additional shape type "vector" to Shape in ModelicaServices

modelica-trac-importer, 2017-01-15:
**Reported by jmattsson on 26 Jan 2016 13:16 UTC**
The class ModelicaServices.Animation.Shape (documented in Modelica.Mechanics.MultiBody.Visualizers.Advanced.Shape) allows a set of predefined shape types (defined in Modelica.Mechanics.MultiBody.Types - including box, sphere, cylinder, etc.). 

We suggest to add an additional shape type "vector". 

 - A vector is a highly relevant shape, in particular for multibody mechanics.     
 - This would allow any post-processing to be handled by the tool. Dymola does that very effectively:
  - In the animation window, one can easily scale the vector, based on the type of quantity (force, torque, velocity, acceleration, angular velocity or angular acceleration, pure distance).
  - Each quantity has an associated color. 
  - A flag can be used to specify if the arrow goes to or from the position.
 - For this shape type, the `extra` parameter would contain the type of quantity. 
 - The shape would have the usual material properties.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1891

beutlich, 2018-04-13:
@jespermattsson I think it would be straight-forward to add "vector" as additional shape type to that list of choices.

https://github.com/modelica/ModelicaStandardLibrary/blob/d49317d3271070e8ce225f3970699f0755be5c99/Modelica/Mechanics/MultiBody/Types.mo#L119-L131

Isn't it a tool issue anyway since choices are only a proposal? Some tools already support this shape for a long time e.g., as packed shape type = 109.

tobolar, 2018-04-13:
I doubt about arrow color being associated with the vector quantity and even being tool dependant. This would IMHO restrict usability of the visualizer and may lead to visualization difficulties for particular user's color palette.
I also guess the arrow diameter should be changable by the user.

beutlich, 2020-01-29:
@HansOlsson Can this be closed after merging #2501?

HansOlsson, 2020-01-30:
> 
> 
> @HansOlsson Can this be closed after merging #2501?

Yes!

closed 2020-01-30 by HansOlsson