modelica/ModelicaStandardLibrary #1898: MSL 3.2.1 build 4 - Fluid pump efficiency

modelica-trac-importer, 2017-01-15:
**Reported by victor&#x2e;avila on 28 Jan 2016 08:56 UTC**
In the model attached, one simple MSL pump and two boundaries were connected to each other.
When I run the simulation in Dymola2016FD01, I got these statistics: 

Translated Model
Constants: 68 scalars
Free parameters: 50 scalars
Parameter depending: 38 scalars
Outputs: 2 scalars
Continuous time states: 2 scalars
Time-varying variables: 24 scalars
Alias variables: 54 scalars
Number of mixed real/discrete systems of equations: 0
Sizes of linear systems of equations: { }
Sizes after manipulation of the linear systems: { }
Sizes of nonlinear systems of equations: {10}
Sizes after manipulation of the nonlinear systems: {2}
Number of numerical Jacobians: 1

Initialization problem
Sizes of nonlinear systems of equations: {10}
Sizes after manipulation of the nonlinear systems: {2}
Number of numerical Jacobians: 1

How can the efficiency of the MSL pump be improved? I mean, getting rid of non-linear equations and jacobian. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1898

modelica-trac-importer, 2017-01-15:
**Modified by dietmarw on 28 Jan 2016 13:56 UTC**

beutlich, 2018-05-29:
@casella I retrieved the original attachment from https://trac.modelica.org/Modelica/attachment/ticket/1898 and attached it above.

casella, 2018-05-29:
I ran the test case with Dymola and I can basically confirm my former diagnosis. The two iteration variables are
- pump.medium.T
- pump.port_a.m_flow

and the two residual equations are the pump head-mass flow relationship and the energy balance

      0 = pump.head-homotopy((pump.N/pump.N_nominal)^2*pump.flowCharacteristic_Unique'"pump"'
        (pump.V_flow_single*pump.N_nominal/pump.N, {1.36E-005, 5.56E-005, 
        0.0001112, 0.0001668, 0.0002224, 0.000278, 0.0003336, 0.0003753}, {
        4.334143, 3.93096389, 3.4404372, 2.855833, 2.3115477, 1.7202237, 
        1.1490489, 0.7525957}), pump.N*(pump.flowCharacteristic_Unique'"pump"'(0,
         {1.36E-005, 5.56E-005, 0.0001112, 0.0001668, 0.0002224, 0.000278, 
        0.0003336, 0.0003753}, {4.334143, 3.93096389, 3.4404372, 2.855833, 
        2.3115477, 1.7202237, 1.1490489, 0.7525957})+pump.V_flow_single*(if abs(
        pump.V_flow_single_init) > 0 then pump.delta_head_init/pump.V_flow_single_init
         else 0))/pump.N_nominal);
      0 = pump.Hb_flow-pump.port_a.m_flow*boundary.ports[1].h_outflow+
        pump.port_a.m_flow*pump.port_a.h_outflow;

I tried setting allowFlowReversal = false in the system object but this didn't help. In principle, once flow reversal has been excluded, it should be possible to compute rho as a function of fixed variables, but apparently the tool cannot figure this out.

You  should contact your tool vendor to discuss why this happens, as well as why a symbolic Jacobian cannot be obtained in this case.

HansOlsson, 2018-05-30:
@casella  wrote "In principle, once flow reversal has been excluded, it should be possible to compute rho as a function of fixed variables, but apparently the tool cannot figure this out."

Note that rho and temperature cannot be directly computed but are part of a non-linear system of equations that depends on the pressure drop (which as default is zero).

For the default configuration (1.01325 bar) the temperature is 20degC and rho is 1.04257216	g/cm3
If boundary1.p is changed to 1.015 bar it seems the temperature becomes	20.00005degC and rho 1.04256718 g/cm3. That system of equations also seems ill-conditioned in various way; and it doesn't seems physical to indirectly determine the temperature in this way for the pump.

HansOlsson, 2018-06-05:
Part of the Jacobian issue is related to the pump-characteristics lacking derivative-information; but there is more to it. 

However, there is more to the Jacobian-issue (have not fully analyzed it) - and I don't see that we should rush it for the coming release.

casella, 2018-06-07:
In fact, if you take into account the enthalpy increase between the inlet and the outlet due to the pump work, it is not odd at all that there is a relationship between dp and T: the higher dp, the higher the fluid enthalpy, the higher the temperature

My 2 cts

casella, 2018-06-08:
If the working fluid is water, the temperature increase is negligible, so one could have a simplified model where Hb_flow is set to zero. 

However, if the working fluid is air, then you get about 1 K for each 1000 Pa of dp. I'm not sure whether this is relevant or not for the typical applications of the library (cooling circuits), @AHaumer, @christiankral can you comment on that?

beutlich, 2019-09-26:
@casella @HansOlsson @christiankral @AHaumer Unfortunately the discussion was stopped after valuable analyses. Could you please decide how to proceed and if MSL v4.0.0 seems possible?

casella, 2019-09-27:
> 
> 
> @casella @HansOlsson @christiankral @AHaumer Unfortunately the discussion was stopped after valuable analyses. Could you please decide how to proceed and if MSL v4.0.0 seems possible?

Do you mean getting this resolved in v4.0.0?

beutlich, 2019-09-27:
> Do you mean getting this resolved in v4.0.0?

Exactly.