modelica/ModelicaStandardLibrary #1899: ModelicaStandardTables_*_read handling is fragile

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 29 Jan 2016 16:05 UTC**
The functions ModelicaStandardTables_\*_read are somewhat fragile in some scenarios. If they are not called for the specific table after the constructor and before the \*_getValue calls it will just silently return 0.0.

It would suggest calling \*_read (with force=false) at the start of \*_getValue, or possibly inline the test to make it more efficient.
I could add it myself if agreed on (would just be a few lines).

Another possibility would just be to generate an error message for this case. I find that less good, but still better than the current variant of silent failure is problematic.

I'm aware of #1103, and I agree that the possibility of re-reading the table is an argument for having it as two functions - but I don't like the implications in terms of error handling.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1899

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 29 Jan 2016 22:59 UTC**
I somehow understand the problem but I do not consider your proposed workaround as clean enough. It was already addressed in #1103 in such a way that dummy input _tableAvailable_ was added to _getTableValue_ to obtain some function call presedence. I wonder why it is considered fragile now and not 3 years ago? Does it help if the _tableAvailable_ input is also passed to the C function \*_getValue? If I remember correctly this was the way how the same issue was resolved for the new Noise functions quite recently. I also might add some additional assert or ModelicaError in \*_getValue if the table data is missing.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 2 Feb 2016 11:20 UTC**
Replying to [comment:1 beutlich]:
> I somehow understand the problem but I do not consider your proposed workaround as clean enough. It was already addressed in #1103 in such a way that dummy input _tableAvailable_ was added to [...]

It was fragile then as well - I just didn't report it, partially because I hadn't fully understood the problem.

The solution can be as simple as (replacing error with some appropriate error message - or just removing that line):
```
double ModelicaStandardTables_CombiTable1D_getValue(void* _tableID, int iCol,
                                                    double u) {
    double y = 0.;
    CombiTable1D* tableID = (CombiTable1D*)_tableID;
    if (tableID != NULL && tableID->table == NULL) {
       /* Ensure that file is read */
       ModelicaStandardTables_CombiTable1D_read(_tableID, 0, 0);
       if (tableID->table == NULL) error
    }
    if (tableID != NULL && tableID->table != NULL && tableID->cols != NULL) {
```

Obviously for all tables.

If writing from scratch I might have used the coding practice of "NULL == ..." instead of "... == NULL".

> Does it help if the _tableAvailable_ input is also passed to the C function \*_getValue?

Unfortunately no.
The problem is if tableAvailable is seen as correctly initialized (e.g. from another call) - but we haven't called the read-routine.

> If I remember correctly this was the way how the same issue was resolved for the new Noise functions quite recently.

I realized that I had to check Noise as well - and that seems ok.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Feb 2016 11:32 UTC**
Replying to [comment:2 hansolsson]:
> Replying to [comment:1 beutlich]:
> > Does it help if the _tableAvailable_ input is also passed to the C function \*_getValue?
> 
> Unfortunately no.
> The problem is if tableAvailable is seen as correctly initialized (e.g. from another call) - but we haven't called the read-routine.

That is what I do not understand. If tableAvailable is seen correctly initialized the table must have been read as expected. We have
```mo
initial algorithm
  tableOnFileRead := readTableData(tableID, false, verboseRead);
equation
  y = getTableValue(tableID, x, y, z, pre(z), tableOnFileRead);
```
thus getTableValue gets called after readTableData. Same approach as in Noise.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Feb 2016 11:47 UTC**
@hansolsson Please add or refer to a test model for easy reproduction.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Feb 2016 13:15 UTC**
 * What do you mean by re-initialization of external objects? Calling the constructor again?
 * Does it help if readTableData is marked impure now? Thus impure functions in initial algorithms shall be called again. Is this specified?

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Feb 2016 13:21 UTC**
 * Does it help if reinit() was used?

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 2 Feb 2016 13:51 UTC**
Replying to [comment:7 beutlich]:
>  \* What do you mean by re-initialization of external objects? Calling the constructor again?

Yes, in order to (re)start the simulation from known values.

>  \* Does it help if readTableData is marked impure now? Thus impure functions in initial algorithms shall be called again. Is this specified?

That is a possibility, but I don't think it is a good idea. 
I can see a case where we just log values to a file and the call in "initial algorithm" clears the file - and one doesn't want to repeat that.

Replying to [comment:8 beutlich]:
>  \* Does it help if reinit() were used?

Don't think so - and that would require an extra state which seems problematic.

Is there a problem with adding the code I indicated above?
I could imagine a tiny issue with efficiency (duplicated null-pointer test), but even that could be avoided.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Feb 2016 14:24 UTC**
Well, I summarize my concern about your proposed C code change again:

I consider it as not clean enough (aka hack) and am rather interested in a generic Modelica solution, that is w/o touching the C files but only the MO files. If you reconstruct an external object for a second simulation taking the initialization from a previous simulation you cannot assume that impure functions on that external object are still valid. This idea of re-initialization a simulation is completely new (at least for me if considering external objects) and should not be resolved en passant by such a hack.

Rather, we should properly discuss and specify how such a problem can be solved in a clean way. Note, there are more open issues on external objects that need to be (well at least) discussed.

Does it help if (protected) _tableOnFileRead_ was not declared as fixed-false-parameter with initial algorithm but as variable instead with a declaration equation? Then it is no longer an initial state and saved in your initialization data. Not sure, if the newly added impure attribute then results in calling _readTableData_ in each simulation step (which essentially matches your above C hack).

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 2 Feb 2016 15:46 UTC**
Replying to [comment:10 beutlich]:
> Well, I summarize my concern about your proposed C code change again:
> 
> I consider it as not clean enough (aka hack) and am rather interested in a generic Modelica solution, that is w/o touching the C files but only the MO files. 

To me the hack is the requirement of calling readTableData() in an initial algorithm instead of doing this in the constructor. 

I view the initial sections are related to initializing the model - not to initializing the objects, and I recall that we at some time replaced "when initial() ..." by external objects in order to make that clearer (see the last paragraph below as well).

In general I prefer that the code is robust (i.e. automatically handling this case correctly) - and as a second option code that reports incorrect use (or sometimes a combination of the two - such that a warning is logged and then the code handles it). 

Currently the code does neither - it detects a problem (null-pointer) and then silently produces a default result - instead of correcting it or reporting it.

> Rather, we should properly discuss and specify how such a problem can be solved in a clean way. Note, there are more open issues on external objects that need to be (well at least) discussed.

I agree that we can consider having some special section for initializing external object - and also need to discuss other problems with them.

> Does it help if (protected) _tableOnFileRead_ was not declared as fixed-false-parameter with initial algorithm but as variable instead with a declaration equation? Then it is no longer an initial [...]

I believe that would be an illegal call of impure function. If tableOnFileRead had been _another external object_ instead it would likely have worked (I haven't tested).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Feb 2016 16:06 UTC**
Replying to [comment:11 hansolsson]:
> Replying to [comment:10 beutlich]:
>
> To me the hack is the requirement of calling readTableData() in an initial algorithm instead of doing this in the constructor.

I guess you read the motivation in #1103.
> 
> I view the initial sections are related to initializing the model - not to initializing the objects, and I recall that we at some time replaced "when initial() ..." by external objects in order to make that clearer (see the last paragraph below as well).

I can't follow here since I am not aware of any other external objects in the MSL. Please give an example where "when initial()" was replaced by an external object.
> In general I prefer that the code is robust (i.e. automatically handling this case correctly) - and as a second option code that reports incorrect use (or sometimes a combination of the two - such [...]

I stated already in comment:1 that adding assert or ModelicaError might be the preferable solution.
> I agree that we can consider having some special section for initializing external object - and also need to discuss other problems with them.

Yes, please.
> I believe that would be an illegal call of impure function. If tableOnFileRead had been _another external object_ instead it would likely have worked (I haven't tested).

This might look like an overkill especially if the table data comes from the model. If it however turns out that having tableOnFileRead as another external object is the clean solution than we are one step further. But I am not sure at all as of now. If this also changes the C API of ModelicaStandardTables we need to re-schedule it for the next MSL anyway (where I already planned to change the API).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Feb 2016 19:14 UTC**
Replying to [comment:11 hansolsson]:
> To me the hack is the requirement of calling readTableData() in an initial algorithm instead of doing this in the constructor. 

If this is now considered as a hack I propose the following list of action items and hope we can discuss and agree on them.

1. Create a new ticket about clarification of initialization of external objects. It should be stated that external objects need solely to be initialized in their constructor, and especially must not be initialized by initial algorithms or initial equations. -> Hans
1. Based on that ticket, discuss external objects at one of the next MDMs and update the MLS (sooner or later). -> MD
1. Ask MAPLib project leader if to the keep milestone of this ticket (#1899) or re-schedule for MSL 3.3. -> Toni
1. For ModelicaStandardTables -> /me
 1. Read table during construction of external object (with force=0). (As already was mentioned in #1103.)
 1. Remove explicit call to _readTableData_ and initial algorithm in MO files. (As already was mentioned in #1103.)
 1. Get rid of dummy input _tableAvailable_ from functions _getTableValue_, _getTableValueNoDer_, _getDerTableValue_ and _getNextTimeEvent_. Since this breaks the backward compatibility it should be re-scheduled for the next major MSL 3.3.
 1. Remove function _readTableData_ from _ModelicaTest.Tables.CombiTable2D_.
 1. Throw an error in C functions _getValue if table data is not available.
 1. Open issue: What to do with _getTableTimeTmin_/_getTableTimeTmax_ and _t_min_/_t_max_ of _CombiTimeTable_? They are pure functions and fixed-false-parameters with initial algorithms, respectively, that depend on the external object.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 3 Feb 2016 09:01 UTC**
After speaking with Martin Otter we spotted two more open issues, esp. for the _CombiTimeTable_.
 7. The current implementation with calling _readTableData_ (with force=false) in the initial algorithm prevents repeated reading of the table data. Assume that reading the table data is an expensive call, then, when moving it to the constructor we get in performance trouble if the constructor/destructor functions are called (in pairs) more than once.
 8. For the _CombiTimeTable_ we call _getNextTimeEvent_ in a _when initial()_ statement. The first call of _getNextTimeEvent_ has side effects on the external object because all the time events are determined and internally stored. Hence this is similar to the _readTableData_ function and should also be avoided (if we forbid impure functions on external objects during initialization).

We also noticed that the same re-initialization problem might also be apparent for imported FMUs.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 5 Feb 2016 17:14 UTC**
Replying to [comment:14 beutlich]:
> After speaking with Martin Otter we spotted two more open issues, esp. for the _CombiTimeTable_.
>  7. The current implementation with calling _readTableData_ (with force=false) in the initial [...]

In case we consider that an important case - and the external object isn't actually used - I would delay loading the table until the first time the table is needed (i.e. proposed variant of getValue - but readTableData is then just an internal function in the C-files).

However, I find it odd that the constructors are called and nothing is used for them.

> 8. For the _CombiTimeTable_ we call _getNextTimeEvent_ in a _when initial()_ statement. The first call of _getNextTimeEvent_ has side effects on the external object because all the time events are [...]

If it is done any time "getNextTimeEvent" is called the first time it is just caching to me and should be safe - but we could move this to the constructor as well.
> We also noticed that the same re-initialization problem might also be apparent for imported FMUs.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 5 Feb 2016 17:18 UTC**
Replying to [comment:13 beutlich]:
> Replying to [comment:11 hansolsson]:
> If this is now considered as a hack I propose the following list of action items and hope we can discuss and agree on them.
>
> 1. Create a new ticket about [...]

Ok, will do.

>  1. Open issue: What to do with _getTableTimeTmin_/_getTableTimeTmax_ and _t_min_/_t_max_ of _CombiTimeTable_? They are pure functions and fixed-false-parameters with initial algorithms, respectively, that depend on the external object.

As I see it that isn't a problem. Similarly as if one called ...getValue in an inital algorithm to perform a table-lookup - it doesn't change the external object; it just stores a value.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 18 Feb 2016 16:01 UTC**
There seems to be the following options, and we should agree on one of the two:

**Constructor reads table**

For Blocks.Tables.XXX, the code would be:
```
protected
  // The table is read from file
  Types.ExternalCombiTable1D tableID = Types.ExternalCombiTable1D(...);
   
equation
  ...
  // Interpolate u[i] in the table
  y[i] = getTableValue(tableID, i, u[i], 1)
```

It is assumed that a Modelica tool calls the constructor only once. If this is not the case, the reading of large tables is possible but will become unnecessarily unefficient.

For Blocks.Sources.CombiTimeTable, the code would be:
```
protected
  // The table is read from file
  Types.ExternalCombiTimeTable1D tableID = Types.ExternalCombiTimeTable1D(...);

initial algorithm
  // Get t_min, t_max
  t_minScaled := getTableTimeTmin(tableID, tableOnFileRead);
  t_maxScaled := getTableTimeTmax(tableID, tableOnFileRead);
  t_min := t_minScaled*timeScale;
  t_max := t_maxScaled*timeScale;
equation
  ...
  when {time >= pre(nextTimeEvent),initial()} then
    nextTimeEventScaled = getNextTimeEvent(tableID, timeScaled, tableOnFileRead);
    if (nextTimeEventScaled < DBL_MAX) then
      nextTimeEvent = nextTimeEventScaled*timeScale;
    else
      nextTimeEvent = DBL_MAX;
    end if;
  end when;

  ...
  // Interpolate u[i] in the table
  y[i] = p_offset[i] + getTableValue(tableID, i, timeScaled, nextTimeEventScaled, pre(nextTimeEventScaled), 1); 
```
In the documentation it is stated, that this table model assumes that after initialization, the table on file is not changed. For example assume that the simulation is terminated at 2 s and the final state stored on file (e.g. to find a steady-state). The next day the simulation is continued started at 2 s and reading the final state at 2 s and starting from this final state. Then, the simulation might fail, if the table on file has been changed when restarting at 2 s (e.g. might be the case if pre(nextTimeEventScaled) is no longer consistent, and/or t_minScaled/t_maxScaled has changed and is used in the user model).

**First access reads table**

For Blocks.Tables.XXX, the code would be:
```
protected
  // The tableID is initialized (a memory for a struct is allocated where the input
  // arguments of the constructor are stored; the address of this struct is returned;)
  Types.ExternalCombiTable1D tableID = Types.ExternalCombiTable1D(...);
   
equation
  ...
  // Interpolate u[i] in the table (if the table is not yet read, it is first read from file
  // and a pointer is stored in the tableID struct, before interpolation takes place)
  y[i] = getTableValue(tableID, i, u[i], 1)
```

This table reading is always efficient because the table is only read once.

For Blocks.Sources.CombiTimeTable, the code would be:
```
protected
  // The tableID is initialized (a memory for a struct is allocated where the input
  // arguments of the constructor are stored; the address of this struct is returned;)
  Types.ExternalCombiTimeTable1D tableID = Types.ExternalCombiTimeTable1D(...);

initial algorithm
  // Get t_min, t_max
  t_minScaled := getTableTimeTmin(tableID, tableOnFileRead);
  t_maxScaled := getTableTimeTmax(tableID, tableOnFileRead);
  t_min := t_minScaled*timeScale;
  t_max := t_maxScaled*timeScale;
equation
  ...
  when {time >= pre(nextTimeEvent), initial()} then
    nextTimeEventScaled = getNextTimeEvent(tableID, timeScaled, tableOnFileRead);
    if (nextTimeEventScaled < DBL_MAX) then
      nextTimeEvent = nextTimeEventScaled*timeScale;
    else
      nextTimeEvent = DBL_MAX;
    end if;
  end when;

  ...
  // Interpolate u[i] in the table (if the table is not yet read, it is first read from file
  // and a pointer is stored in the tableID struct, before interpolation takes place)
  y[i] = p_offset[i] + getTableValue(tableID, i, timeScaled, nextTimeEventScaled, pre(nextTimeEventScaled), 1); 
```

Again, in the documentation it is stated, that this table model assumes that after initialization, the table on file is not changed. For example assume that the simulation is terminated at 2 s and the final state stored on file (e.g. to find a steady-state). The next day the simulation is continued started at 2 s and reading the final state at 2 s and starting from this final state. Then, the simulation might fail, if the table on file has been changed when restarting at 2 s (e.g. might be the case if pre(nextTimeEventScaled) is no longer consistent, and/or t_minScaled/t_maxScaled has changed and is used in the user model).

There seems to be no essential difference for the two variants and therefore it is probably cleaner to use variant 1.

Note, there is no way with the current design that CombiTimeTable works correctly, if the table is modified before re-reading it af the previous final state, because t_min/t_max are parameters and are read from the final state, and becuase pre(nextTimeEvent) is also read from the final state and cannot be set to a correct value when the simulation is restarted.

Note, we should also simplify the MSL, by introducing the package "Blocks.Tables.Internal" which shall contain all external function calls to the tables, in order that these definitions do not have to be duplicated several times. I can do this, if you (Thomas) agree.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 18 Feb 2016 19:20 UTC**
Shouldn't we remove the initial algorithm completely, i.e. have
```mo
    final parameter Modelica.SIunits.Time t_min=t_minScaled*timeScale
      "Minimum abscissa value defined in table";
    final parameter Modelica.SIunits.Time t_max=t_maxScaled*timeScale
      "Maximum abscissa value defined in table";
    final parameter Real t_minScaled=getTableTimeTmin(tableID, tableOnFileRead)
      "Minimum (scaled) abscissa value defined in table";
    final parameter Real t_maxScaled=getTableTimeTmax(tableID, tableOnFileRead)
      "Maximum (scaled) abscissa value defined in table";
```
instead of 
```mo
    final parameter Modelica.SIunits.Time t_min(fixed=false)
      "Minimum abscissa value defined in table";
    final parameter Modelica.SIunits.Time t_max(fixed=false)
      "Maximum abscissa value defined in table";
    final parameter Real t_minScaled(fixed=false)
      "Minimum (scaled) abscissa value defined in table";
    final parameter Real t_maxScaled(fixed=false)
      "Maximum (scaled) abscissa value defined in table";
/*...*/
  initial algorithm
    t_minScaled := getTableTimeTmin(tableID, tableOnFileRead);
    t_maxScaled := getTableTimeTmax(tableID, tableOnFileRead);
    t_min := t_minScaled*timeScale;
    t_max := t_maxScaled*timeScale;
```
as otherwise a reinitialized simulation relies on wrong t_min/t_max from the previous simulation and the strong requirement that the external table file must not change?

Since tableOnFileRead becomes obsolete now it would be smarter to postpone this issue to the next MSL update where I am allowed to remove it.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 19 Feb 2016 08:32 UTC**
Replying to [comment:18 beutlich]:
> Shouldn't we remove the initial algorithm completely, i.e. have

I agree that it seems logical, but would delay it to the next release.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 19 Feb 2016 08:46 UTC**
Replying to [comment:18 beutlich]:
> Shouldn't we remove the initial algorithm completely, i.e. have
> ```mo
>     final parameter Modelica.SIunits.Time t_min=t_minScaled\*timeScale
>       "Minimum abscissa value defined in table"; [...]

Yes, you are completely right (remove initial algorithm section and call the function in the parameter declaration).

> 
> Since tableOnFileRead becomes obsolete now it would be smarter to postpone this issue to the next MSL update where I am allowed to remove it.

Hm. Do you mean to keep the tables as they are now, and do not perform the changes above?

The trouble is that no one knows when an MSL version will be released that is no longer backwards compatible (of course we try to keep MSL backwards compatible as long as possible). Therefore, we should not wait until this event will happen because with luck this will be in 5-10 years. The table functions are anyway not really critical, because until now they are only interfaced in a protected section and are not directly accessible.

Ths simplest solution is just to keep all interfaces, and also keep function tableOnFileRead (if this function is used by a user, the only drawback is that the file is read twice; but it still works).

modelica-trac-importer, 2017-01-15:
**Comment by anonymous on 19 Feb 2016 08:53 UTC**
Replying to [comment:6 hansolsson]:
> External objects must be re-initialized in that case (since we don't have a save/restore interface) - but we shouldn't initialize everything according to the initial algorithms.

So introduce save/restore to specification 3.4 instead, and allow its use in MSL 3.4. If Modelica does not support save/restore it cannot be important to fix it in the current MSL and current specification.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 19 Feb 2016 08:58 UTC**
Replying to [comment:21 anonymous]:
> Replying to [comment:6 hansolsson]:
>
> So introduce save/restore to specification 3.4 instead, and allow its use in MSL 3.4. If Modelica does not support save/restore it cannot be important to fix it [...]

This is a timing issue: Releasing Modelica 3.4 takes time, until all the Modelica tools supports it takes A LOT OF TIME (only Dymola supports 3.3, and 3.3 was released 2012), and only afterwards one could utilize this feature in MSL. To summarize, it is out of the question to wait for it.

With the slight changes above, it will be possible that a model containing tables can be restarted from a previous simulation run and for some end-users this is important (independent of the tool)

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 19 Feb 2016 09:01 UTC**
Replying to [comment:21 anonymous]:
> Replying to [comment:6 hansolsson]:
>
> So introduce save/restore to specification 3.4 instead, and allow its use in MSL 3.4. If Modelica does not support save/restore it cannot be important to fix it [...]

Save/restore is important for external objects **that have an internal state that influences the simulation**, and we need to work on that since there are important applications.

However, tables only have an internal state for efficiency reasons - it shouldn't impact the result (as far as I recall - there might be some minor exception). 
In theory we could skip the table-object and just read the table file for each table-lookup - and we would get the same result, just slower.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 19 Feb 2016 09:06 UTC**
If the issue is "too much changes to the table implementation" now, we could also follow the original suggestion by Hans and only make a  minor change: If the table is not there when interpolating, it is just read. As I recall, this would be only a one-line change in the table (I and also do it a and test it, if timing is the issue)

modelica-trac-importer, 2017-01-15:
**Comment by otter on 19 Feb 2016 14:36 UTC**
Yet another possibility is to keep all table C-functions unchanged, but
- Rename the constructor C-function to ModelicaStandardTables_CombiTimeTable_initWithReadFile (this function just calls the existing functions to init and read the file).
- Change the Modelica code according to the above suggestions (I talked to Hans, and it is fine for him to initialize the T_min/T_max parameters in the declaration as suggested above).

This looks to me like a simple, clean, and backwards compatible solution.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 19 Feb 2016 15:33 UTC**
As proposed, moved external C-function definitions of tables out of the protected sections in a separate package (Blocks.Tables.Internal) in b6c30ee8ce312c83ab48bbc89a9bdd507481793b.
Advantages:
- No double definitions anymore (and therefore easier to maintain).
- Code for the table models is simpler (because function definitions no longer present in the protectec section).
- The functions can also be called in another context (e.g. for scripting).
Tested the changes with "Check" and "Check with Simulations".

modelica-trac-importer, 2017-01-15:
**Comment by otter on 22 Feb 2016 16:41 UTC**
The proposed changes are in the attached patch-file. Thomas, please apply this patch and re-generate the table object libraries. I have tested with ModelicaTest (by including the new C-code with an Include; after the test, I have removed the Include directive and included the code in ModelicaStandardTables.c).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 24 Feb 2016 13:54 UTC**
Two remarks:
 * You changed the type of the external object _tableID_ to parameter? Was this by design? I guess it is not needed in the context of this ticket.
 * The constructor of the external object now takes the parameter _verboseRead_ as input argument. However, _verboseRead_ is disabled (in the dialog) if the table is not to be read from file. Thus, it is not a clean implementation. In the current implementation _verboseRead_ is only accessed if _tableOnFile_ is true.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 24 Feb 2016 14:12 UTC**
Replying to [comment:28 beutlich]:
> Two remarks:
>  \* You changed the type of the external object _tableID_ to parameter? Was this by design? I guess it is not needed in the context of this ticket.

You are right. It does not matter, and it is probably safer to not use parameter (so please, remove it).

> \* The constructor of the external object now takes the parameter _verboseRead_ as input argument. However, _verboseRead_ is disabled (in the dialog) if the table is not to be read from file. Thus, [...]

This approach is used in many Modelica models and in many (non-Modelica related) libraries: Some parameters only have an effect if some other parameters have a particular value (otherwise their value is ignored). Just take the LAPACK library, where this is used everywhere. In Modelica this is anyway just a user-interface issue define via the Dialog annotation (so that a tool can lock a dialog field). Still, the value can still be changed in the text layer (even if the parameter is not used).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 24 Feb 2016 14:18 UTC**
Replying to [comment:29 otter]:
> Replying to [comment:28 beutlich]:
>
> You are right. It does not matter, and it is probably safer to not use parameter (so please, remove it).

OK.
> This approach is used in many Modelica models and in many (non-Modelica related) libraries: Some parameters only have an effect if some other parameters have a particular value (otherwise their [...]

So no need to have it like this: if tableOnFile then verboseRead else false

modelica-trac-importer, 2017-01-15:
**Comment by otter on 24 Feb 2016 14:23 UTC**
Replying to [comment:30 beutlich]:
> Replying to [comment:29 otter]:
> OK.
> So no need to have it like this: if tableOnFile then verboseRead else false

Hm. Do you mean to have "if tableOnFile then verboseRead else false" as actual argument for verboseRead to the C-function (constructor)? Feel free to use either solution. Providing directly parameter "verboseRead" is also fine (since it does not have an effect).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 25 Feb 2016 12:21 UTC**
One more problem with the proposed initWithRead shortcut constructors is, that they leak memory if reading of the table fails, i.e. ModelicaError is called in the read function or its called functions.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 25 Feb 2016 13:20 UTC**
Thomas, since it seems that you do not have time to fix this, I will do it tomorrow morning (based on the proposal by Hans).

modelica-trac-importer, 2017-01-15:
**Comment by otter on 25 Feb 2016 13:22 UTC**
Replying to [comment:34 otter]:
> Thomas, since it seems that you do not have time to fix this, I will do it tomorrow morning (based on the proposal by Hans). 

Forgot: This includes regeneration of the table object libraries (so you do not have to do anything).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 26 Feb 2016 07:34 UTC**
For the record: #1907 is the corresponding MLS ticket on improvement of the initialization of external objects.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 26 Feb 2016 09:38 UTC**
Replying to [comment:30 beutlich]:
> Replying to [comment:29 otter]:
> OK.

I recognized that this is not possible in Dymola (and most likely this is also the Modelica Specification interpretation). The reason is:
```
  final parameter Real t_minScaled = Internal.getTimeTableTmin(tableID);
  final parameter Modelica.SIunits.Time t_min = t_minScaled*timeScale;
protected 
  parameter Modelica.Blocks.Types.ExternalCombiTimeTable tableID = ...
```
tableID is used in the parameter expression t_minScaled, which is in turn used in the parameter expression t_min. According to the Modelica Specification this means that all variables referenced in the parameter expression must be parameters or constants. Without the "parameter" prefix for tableID, Dymola treats tableID not as parameter and triggers an error (when the "parameter" prefix is not present. I therefore keep this implementation.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 26 Feb 2016 12:06 UTC**
Fixed in f85205a60371a2ebd01d47d2232808e9b814c85b along the lines of the discussion above (new constructor functions that initialize tableID and read the file in the constructor). Checked with Dymola 2017 Alpha for win64 by running Check and CheckWithSimulation for ModelicaTest.Tables.

modelica-trac-importer, 2017-01-15:
**Changelog modified by otter on 26 Feb 2016 12:06 UTC**
Initialization of Blocks.Tables.XXX and Blocks.Sources.CombiTimeTable changed, so that the tables are read from file in the constructor. It is then possible to stop a simulation and continue simulation also if tables are contained in the model

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 29 Feb 2016 22:14 UTC**
I'd like to
 1. rename _initWithRead to init2
 2. add the missing extrapolation parameter as a stub to the 1D and 2D variants in preparation for #1839
 3. fix the memory leaks if the constructor fails in read, isValidTable or getTableSource.

However, the third action item requires some thorough rework and test which I only can do during spare-time.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 1 Mar 2016 07:18 UTC**
Replying to [comment:39 beutlich]:
> I'd like to
>  1. rename _initWithRead to init2

The "old name" is better, since clearer. Please, keep it.

>  2. add the missing extrapolation parameter as a stub to the 1D and 2D variants in preparation for #1839

And a stub for "instance name", and a stub for uniform-grid, and a stub for an interpolation method that requires additional options, ...
Please, don't do this now. New functions should be introduced, when there is a design for it and by experience the chance is high that the design is wrong (just by guessing, if you don't implement and test it).

>  3. fix the memory leaks if the constructor fails in read, isValidTable or getTableSource.

Explain why there should be a memory leak (if there is one, it was already present in the previous release). Note, the current implementation is
```
void* ModelicaStandardTables_CombiTable1D_initWithRead(...) {
   void* tableID = ModelicaStandardTables_CombiTable1D_init(...);
   double success; 
   if ( tableID == NULL ) return NULL;                                                         
   success = ModelicaStandardTables_CombiTable1D_read(tableID, 0, verbose);
   if ( success == 0.0 ) {
      ModelicaStandardTables_CombiTable1D_close(tableID);
      return NULL;
   }
   return tableID;
}
```
Note, tableID <> NULL is only returned if memory is successfully allocated.
If xxx_CombiTable1D_read(..) has an error during reading, it should return "success = 0.0" according to the documentation, but then the potentially allocated memory is freed.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Mar 2016 12:31 UTC**
Reopen now and give detailed reasons later, i.e., in my Modelica dedicated spare-time. Cannot afford to write hour-long reasoning during office hours.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Mar 2016 12:32 UTC**
If I accept a ticket it means I am going to work on it. So please do not close it instantly.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Mar 2016 17:25 UTC**
This is why I have concerns with the current solution:

 * Both in comment 3 of #1820 and comment 11 of #1839 I stated that I do not want to introduce new C-API functions in ModelicaStandardTables.h. Whereas #1820 would need an additional verbose parameter for the constructors, #1839 requires a new extrapolation parameter (also for the constructors). By purpose, I decided to not touch the C-API and in case of #1839 reschedule the ticket milestone.
 * With this ticket (#1899) a new use case came around, which is the reinitialization of a saved simulation with external objects. Note that since #1103 there was not a single issue with the existing ModelicaStandardTables regarding reading tables from files outside the constructor but in an extra function called during an initial algorithm section. I understand the new problem, but see, we needed to touch the MLS for it (#1907).
 * Reinitialization of a saved simulation with external objects is a completely new topic. This ticket (#1899) showed some ideas/questions if we shall add new API functions to ExternalObject which make save and restore easier.
 * If we fix it now as hack by introducing new C-API functions (for the verbose parameter of the constructor) though neither #1907 nor the ExternalObject is properly dicussed we run the risk of adding stuff that is hard to get undone later.
> New functions should be introduced, when there is a design for it and by experience the chance is high that the design is wrong (just by guessing, if you don't implement and test it).
 * You are right here. If there is neither a design nor a proper discusssion on it we should not add new C-API functions _initWithRead. Despite the missing design, the name _initWithRead is also misleading if the table data is not read from file but taken from the model. _initWithReadIfTableOnFile does not really sound better, thus my generic proposal _init2.
 * About the unlucky memory leak: Before f85205a60371a2ebd01d47d2232808e9b814c85b the following flow of functions worked:
   1. Call constructor w/o reading table data
   1. Try to read table data from file
   1. Reading table data failed as e.g., MAT file format v7.3 is not supported by the tool. Read function fails with ModelicaError
   1. Call destructor to free all allocated memory
 * After f85205a60371a2ebd01d47d2232808e9b814c85b we now always have:
   1. Call constructor and try to read table data from file
   1. Reading table data failed as e.g., MAT file format v7.3 is not supported by the tool. Read function fails with ModelicaError and does not return to the calling constructor and leakes resources.
   1. No need/way to call destructor
 * You see, it was a design decision to not call the read function in the constructor. If this design decision is now reverted (by validity of #1907) we also need to update the C code properly. The current C implementation punishes the hack of f85205a60371a2ebd01d47d2232808e9b814c85b unexpectedly but dramatically. For sure it can and needs to be solved in an clean way.
 * Please note, that SimulationX always runs shared objects for the simulation which is the reason I do care a lot for resource leaks. In case of Dymola, memory leaks are not relevant as long as the executable binary runs in an own process.

My proposal:
 * Revert f85205a60371a2ebd01d47d2232808e9b814c85b and keep particularly ModelicaStandardTables.h untouched
 * Rethink of a proper solution of how to deal with the reinitialization of external objects (not yet considered as part of #1907). Maybe DS Lund can present what they are actually doing with some more technical details.
 * Reschedule this ticket (#1899) to the next MSL
 * Solve it in a really clean way along with other relevant table proposals (from #1153)

Since I feel this is a very important issue the MAP-LIB members should vote on this (or an alternative) proposal.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 3 Mar 2016 04:58 UTC**
Reverted 
 * f85205a60371a2ebd01d47d2232808e9b814c85b, so tables are again read in the initial section and not in the constructor.
 * b6c30ee8ce312c83ab48bbc89a9bdd507481793b, so table functions are again protected functions in the model and cannot be accessed, say, from scripting. This reversion is necessary because otherwise functions would be present in MSL that do not return in case of error.

These reversions have been tested with Dymola 2017 Beta.1 (Check and CheckWithSimulation on Modelica and ModelicaTest).

Opened a branch in r9121 (https://svn.modelica.org/projects/Modelica/branches/development/ModelicaStandardTables-Fix1899), in which the issue is fixed with the suggestion from Hans (= if table is not present in xx_getValue(..), then the table is again read from file). This is discussed in the separate ticket #1941. A tool vendor might use the ModelicaStandardTables.lib/.a file from this branch in order to get an immediate fix for the bug reported in #1899.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 3 Mar 2016 04:59 UTC**
The reversions have been done in bc3041acd345cc9072a0caddfd149f267eab5e40.

modelica-trac-importer, 2017-01-15:
**Comment by sjoelund&#46;se on 3 Mar 2016 16:37 UTC**
I added #1950 to propose adding a save/restore interface

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 5 Apr 2016 08:17 UTC**

beutlich, 2017-04-10:
I am going to work on this (previously heavily discussed) issue now and plan to fix it according to the items listed in https://github.com/modelica/Modelica/issues/1899#issuecomment-272680317 and also the improvements proposed by @MartinOtter, i.e, Modelica.Blocks.Tables.Internal.

There are two issues I'd like to discuss with @HansOlsson
* If we move getTableTimeTmin/getTableTimeTmax out of the initial algorithm (as proposed by https://github.com/modelica/Modelica/issues/1899#issuecomment-272680356) we need to change the external object tableID from variable to parameter. Do you see any isse with this change?
* If we move getNextTimeEvent to be called from the ctor of the external object (as proposed by https://github.com/modelica/Modelica/issues/1899#issuecomment-272680329) can we savely drop the when initial() case? Also we need to pass time to the ctor then.

beutlich, 2017-04-12:
@HansOlsson I have my patch ready now - except the second issue if getNextTimeEvent ( = ModelicaStandardTables_CombiTimeTable_nextTimeEvent) should also be called from the constructor, such that we can drop the when initial() case. OK for you?

beutlich, 2017-04-12:
Yeah, finally closing the most commented (open) issue.  :satisfied: Get the binaries from:
* [win32.zip](https://github.com/modelica/Modelica/files/917913/win32.zip)
* [win64.zip](https://github.com/modelica/Modelica/files/917914/win64.zip)
* [linux32.tar.xz](https://bitbucket.org/tbeu/downloads/downloads/ModelicaStandardTables_linux32.tar.xz)
* [linux64.tar.xz](https://bitbucket.org/tbeu/downloads/downloads/ModelicaStandardTables_linux64.tar.xz)

closed 2017-04-12 by beutlich