modelica/ModelicaStandardLibrary #1900: Clean up of sensitive test for Modelica.Fluid

modelica-trac-importer, 2017-01-15:
**Reported by hubertus on 29 Jan 2016 16:21 UTC**
The test model ModelicaTest.Fluid.TestExamplesVariants.BranchingDynamicPipes_MomentumSteadyState worked in earlier Dymola releases, but failed unless tolerances were tightened considerably in the latest version. This could be because it is numerically too sensitive to be a good test model. It is suggested that the Library officer looks over the test. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1900

modelica-trac-importer, 2017-01-15:
**Comment by otter on 19 Feb 2016 13:42 UTC**
Added Tolerance=1e-5 annotation in 563e24c54b18f9a8cf2991effbd41233bd6b560a, in order that this model simulates in Dymola 2017 Alpha.

casella, 2018-05-29:
With the current tolerance setting (which is standardized), the model works both in Dymola and OpenModelica. I guess these models are numerically sensitive by their very nature, since distributed friction is quite low compared to absolute pressures. 

I would attempt a structural fix (e.g. using delta-p as preferred states) if the issue was severe, but I reckon a tolerance of 1e-5 for this kind of models is just fine.

closed 2018-05-29 by casella