modelica/ModelicaStandardLibrary #1904: Support for textures in ModelicaServices

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 8 Feb 2016 16:47 UTC**
In several applications it is useful to have specific textures for the animation objects (I will only propose it for surfaces for now - but it could be generalized to some other shapes as well), and Modelon asked me to propose adding this to ModelicaServices.

A simple use case is to visualize a simple road for vehicle simulations. For surfaces one could imagine handling this by having a multicolored surfaceCharacteristic that returns the texture-colors, but a more direct approach seems better in my opinion.

For that to work the following two parts are needed:
* Specifying which texture to use (and default to no texture).
* Mapping surface coordinates to texture coordinates.

Dymola supports specifying the texture as an URL (and also as a "texture number 1-99" - which seems too awkward to standardize).

For the texture itself it is a matter of extending Modelica.Utilities.Internal.PartialModelicaServices.Animation.PartialSurface with:
```
  parameter String texture="" 
    "URI for texture - or empty if no texture";
```
\--
The default texture coordinates are something like: (u/nu, v/nv).

The important non-trivial mapping is that the texture is shorter than the surface and thus replicated. What we currently have are two possibilities:
* The texture is replicated nreplicate_u times in the u-direction and nreplicate_v times in the v-direction.
* The length of the texture is given in meters in the u-direction (think of a small road-section duplicated many times).

The following parameters would support that:
```
  parameter Boolean useLengthInMeters=false;
  parameter Integer replicate_u=1 annotation(Dialog(enable=not useLengthInMeters));
  parameter Integer replicate_v=1;
  parameter Real lengthInMeters annotation(Dialog(enable= useLengthInMeters));
```
\--
Obviously the exact parameter names and parameters can be modified.

Generalizing this to other shapes (like spheres, boxes, ... - but not dxf-files) would be possible. However, I view it as lower priority and note that we would need to standardize texture coordinates in more detail (especially for shapes that have multiple "surfaces").

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1904

modelica-trac-importer, 2017-01-15:
**Modified by jmattsson on 18 Feb 2016 08:44 UTC**

HansOlsson, 2018-01-15:
The proposal above is straightforward, and I could create a pull-request based on this idea of having textures for shapes in ModelicaServices - if there is interest.

(I was reviewing assigned issues.)