modelica/ModelicaStandardLibrary #1905: Add icon for ExternalObject

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 9 Feb 2016 15:11 UTC**
Would be nice if there was a default icon for type ExternalObject such that Modelica.Blocks.Types.ExternalCombiTimeTable (and other) can extend from it.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1905

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 9 Feb 2016 15:33 UTC**
Assigning to dhedberg who cared about #1096.

beutlich, 2017-03-24:
@d-hedberg Do you have any nice icon idea?

HansOlsson, 2017-03-27:
I don't think we should focus on this, since:

- Each class derived from ExternalObject will be different (an external object for a table and an external object used to communicate with an input/output device like a joystick).
- You are not supposed to drag'n'drop and graphically connect to the ExternalObject.

beutlich, 2017-03-27:
Yes, I also can live with it.

closed 2017-03-27 by beutlich