modelica/ModelicaStandardLibrary #1906: Proposal to hide non-graphical protected outer world objects

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 12 Feb 2016 10:11 UTC**
Many classes in Modelica.Mechanics.MultiBody e.g. Modelica.Mechanics.MultiBody.Joints.Revolute have a non-graphical protected outer world object, and indirectly in Modelica.Mechanics.MultiBody.Interfaces.PartialTwoFrames that have outer world that is used in derived classes in the same way.

This non-graphical model-component world is only used for getting parameter-values, but prevents using the joint as a base-class and adding "world" or simulating it on its own.

It might be good for some cases - but there is another possibility that we can use that avoids any local name clash.

Replace: 
```
protected
  outer Modelica.Mechanics.MultiBody.World world;
  Visualizers.Advanced.Shape cylinder(...) if world.enableAnimation and animation;
```
by 
```
protected
  Hide hide; 
  Visualizers.Advanced.Shape cylinder(...) if   hide.world.enableAnimation and animation;
```
with:
```
  model Hide
    outer Modelica.Mechanics.MultiBody.World world;
  end Hide;
```
all components can now have graphical annotations - and obviously better names could be used, and Hide is a common class that should be reused. 

The difference is that the class no longer has an "protected outer world".

Having some syntactic sugar to hide "hide" would be possible - as would be special rules for inheritance and overriding, but I think that would be too large changes for too little gain.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1906

beutlich, 2018-05-08:
> and obviously better names could be used, and Hide is a common class that should be reused.

What about Modelica.Mechanics.MultiBody.Internal.WorldWrapper, though Internal is not yet there?

HansOlsson, 2018-05-09:
> What about Modelica.Mechanics.MultiBody.Internal.WorldWrapper, though Internal is not yet there?

Something like that seems like a good descriptive name.

HansOlsson, 2018-05-14:
I began doing this, and realized that this will be a change of protected elements of Interface-models, and e.g. if someone made their own copy of the Revolute-joint the change will impact that model.

Thus it seems we need a conversion script for this; and move it to the next milestone.

beutlich, 2019-10-23:
> 
> 
> I began doing this (in HansOlsson:WorldWrapper ), and realized that this will be a change of protected elements of Interface-models, and e.g. if someone made their own copy of the Revolute-joint the change will impact that model.
> 
> Thus it seems we need a conversion script for this; and move it to the next milestone.

@HansOlsson Since MSL v4.0.0 comes with a conversion script, could you please continue the started work. Thanks a lot.