modelica/ModelicaStandardLibrary #1909: Functions missing derivative functions, that are already pressent

modelica-trac-importer, 2017-01-15:
**Reported by veje&#x2e;johannes on 16 Feb 2016 10:04 UTC**
In Media.IdealGases.Common.Functions there are three functions where the corresponding derivative function is available but not implemented. 

These three functions are; h_T, h_Tlow, and s0_Tlow with the corrosponding functions: h_T_der, h_Tlow_der, and s0_Tlow_der.

```
function h_T 
  ...
  annotation (Inline=false,smoothOrder=2);
end h_T;
```

For a fourth function, cp_Tlow, the derivative function is implemented.

```
function cp_Tlow 
  ...
  annotation (Inline=false, derivative(zeroDerivative=data) = cp_Tlow_der);
end cp_Tlow;
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1909

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 16 Feb 2016 14:35 UTC**
There are two options for getting derivatives of functions:
 1. the developer writes a derivative function and uses the derivative-annotation or
 1. the tool computes the derivative automatically through AD (automatic differentiation) based on the smoothOrder annotation given by the developer. 

I have chosen the latter, and I think most tools are fine with that. Note that one has to do either one or the other. But: this code was written when Dymola was the only tool that supported the Media library. So I'd like to use this ticket to get input from all tool developers:
 - do you support computation of derivatives based on the SmoothOrder annotation? 
 - which of the two options do you prefer? 

Note that the derivative functions are historical, Dymola supported that before supporting SmoothOrder.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 16 Feb 2016 14:46 UTC**
For the record: SimulationX supports both options.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 17 Feb 2016 16:37 UTC**
Dymola also supports both - and I see several reasons to prefer smoothOrder:
* Reducing the risk of errors by having less code
* Declarative - by stating that the function is 2 times continuously differentiable instead of giving a derivative

The cases where a function has a smooth derivative, but it cannot be computed using automatic differentiation of the algorithm as it is written is pretty rare (but I am sure some have found those cases).

modelica-trac-importer, 2017-01-15:
**Comment by jmattsson on 18 Feb 2016 08:25 UTC**
JModelica.org supports both.

beutlich, 2019-05-07:
As again pointed here by @thorade the derivative or smoothOrder, respectively annotation is missing for s0_Tlow.

PR #2919 (2019-07-26): Fix algorithm of IdealGases.Common.Functions.s0_Tlow_der
@finback-at @thorade FYI

Fix #2870, fix #1909.
closed 2019-07-26 by beutlich