modelica/ModelicaStandardLibrary #1910: Issue with uninitialized record fields

modelica-trac-importer, 2017-01-15:
**Reported by awittkop on 16 Feb 2016 16:33 UTC**
In the MSL file Media/Water/IF97_Utilities.mo on lines 894 and 931 in the functions boilingcurve_p and dewcurve_p respectively, a property record (Common.IF97PhaseBoundaryProperties) is populated.

For both these functions, in the else clauses of the contained if statement, the cp, vt, and vp fields are left uninitialized.

Isn't it necessary that (all parts of) declared outputs be set before returning?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1910

modelica-trac-importer, 2017-01-15:
**Modified by dietmarw on 17 Feb 2016 07:31 UTC**

thorade, 2020-02-10:
Here is direct link to the code:
https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Media/Water/IF97_Utilities.mo#L894-L966

thorade, 2020-02-14:
These records are to my understanding only used for some iterations, not to be used by end users (these functions are not part of the interface). Thetwo branches of the if-else are different, because in the two different regions different iterations are used. Calculating values that are not used later on, might be a waste of time, but we could of course do it.

If we just want to comply with the language specification, we could just explicitly set each missing value to 0.