modelica/ModelicaStandardLibrary #1913: Modelica.Media.IdealGases.Common.MixtureGasNasa.specificEntropy correct?

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 17 Feb 2016 16:46 UTC**
Based on #1909 I tried to see if functions are would be complicated for automatic differentiation (or algorithmic differentiation for those that prefer that term), and found something else that is suspicious:

Modelica.Media.IdealGases.Common.MixtureGasNasa.specificEntropy

```
s :=  s_TX(state.T, state.X) - sum(state.X[i]*Modelica.Constants.R/MMX[i]*
    (if state.X[i]<Modelica.Constants.eps then Y[i] else 
    Modelica.Math.log(Y[i]*state.p/reference_p)) for i in 1:nX);
```
The handling of low concentrations look odd (switching between Y and log (Y\*...), and at least there should be a comment explaining how it works.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1913

modelica-trac-importer, 2017-01-15:
**Modified by dietmarw on 17 Feb 2016 17:45 UTC**

beutlich, 2017-08-21:
@hubertus65 Any comment on this question?

hubertus65, 2018-01-20:
The log(Y) term goes haywire when Y[i] goes to 0, and the term is actually 0 for Y[i] equal to zero. The existing solution is basically the most trivial way to protect against that singularity. There are many instances when a gas mixture is used when a component should be actually 0, which would result in a NaN. There may be better ways to protect this numerically, but I have not seen the need. I am happy to accept a better solution if someone knows one. I know this is a discontinuity, but triggering it in practice is rare.

hubertus65, 2018-01-21:
I changed this from bug to enhancement: this is correct in the two main use cases considered: small concentrations, and components that are included in a stream for completeness but are zero at a given part of the system. In the 18 years this code has been around in total, I have not seen actual problem reported due to this. Hans: if you have a good idea for smoothing the transition, I am happy to review that.

HansOlsson, 2018-02-16:
The basic problem I have is that the formula is mixing different variables and thus it isn't clear that small for one is small for another. By looking inside massToMoleFractions I believe we could reformulate the sum-term from:
```
sum(state.X[i]*Modelica.Constants.R/MMX[i]*
    (if state.X[i]<Modelica.Constants.eps then Y[i] else 
    Modelica.Math.log(Y[i]*state.p/reference_p)) for i in 1:nX)
```

beutlich, 2019-10-23:
@hubertus65 @thorade Can you please comment on @HansOlsson's analysis? Thanks a lot!

thorade, 2020-02-14:
Each steps seems to be correct, only the first step goes from /MMX to *MMX and I don't see why.
If MassFractions X are low, MoleFractions Y are also low, so using either one as switching criterion seems to be OK as well.

massToMoleFractions 
https://github.com/modelica/ModelicaStandardLibrary/blob/cf26871eee2122b86641aed0c03bdd0c7ba8d6fa/Modelica/Media/package.mo#L5121-L5139