modelica/ModelicaStandardLibrary #1918: re/im in Complex are not really replaceable

modelica-trac-importer, 2017-01-15:
**Reported by choeger on 22 Feb 2016 13:38 UTC**
Although I do not know the motivation for making these fields replaceable, they are clearly intended to be redeclared at some point. The operator functions however make this pretty much impossible, as they import the record `Complex` from the **toplevel**. 

See e.g. https://trac.modelica.org/Modelica/browser/Modelica/trunk/Complex.mo#L45

These import statements seem to serve no other purpose than to explicitly avoid redeclarations of re/im, so one of the two is redundant and confusing.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1918

AHaumer, 2018-01-04:
I think this stems from the fact that the operators are encapsulated. 
Maybe people with deeper insight into the language comment on that.

HansOlsson, 2018-01-05:
Considering that `Modelica.SIunits.Complex*` (including ComplexCurrent, ComplexVoltage etc) actually redeclares re/im in a correct way I would say that ticket should be closed as invalid.

Obviously that is limited to scalar subtypes of real - and the operations do not use the sub-types. But that doesn't matter - since the goal of redeclaring re/im was just to change the unit for the variable.
(And there are also language reasons why ComplexCurrent.'+' should be the same as Complex.'+').

This is different from a template in e.g. C++ that can generate `complex<float>` and `complex<long double>` that are not sub-types of each other (I don't know if you can use it for `complex<matrix<double>>`).

closed 2018-01-05 by AHaumer

beutlich, 2018-01-05:
FYI @choeger.