modelica/ModelicaStandardLibrary #1922: Wrong values in some ideal gases (HO2 and HO2minus)

modelica-trac-importer, 2017-01-15:
**Reported by veje&#x2e;johannes on 24 Feb 2016 11:44 UTC**
I noticed numerical errors in Media.IdealGases.Common.SingleGasesData. At least two of the data records are wrong, these are "HO2" and "HO2minus". The errors are in "H0", "blow", "ahigh", and "bhigh".

I believe the reason for the errors are found in the way the data is listed in the database (http://www.grc.nasa.gov/WWW/CEAWeb/). For each of the species there are a number of rows and one of these rows (row 5) contains the last two polynomial coefficients (column 1 to 32) and the two integration constants (column 49 to 80). Column 33 to 48 are left blank. However, for "HO2" and "HO2minus" column 33 to 48 are not blank, but contains the value 0.000000000D+00. 

Row 5 from random species
```
 1.713797538D-08-4.551281870D-12                -1.631720064D+04 3.645052500D+01
```

Row 5 from "HO2"
```
 1.895600056D-08-4.828567390D-12 0.000000000D+00-5.873350960D+03 5.193602140D+01
```

This could cause an automated process to shift the coefficients by one place, which is what appears to have happened.

Values of "ahigh" for "HO2" as they are in package
```
ahigh={10002.162,
       -1810669.724,
       4963.19203,
       -1.039498992,
       0.004560148530000001,
       -1.061859447e-006,
       1.144567878e-010}
```
   
Values of "ahigh" for "HO2" as they are in the database
```
ahigh={-1810669.724,
       4963.19203,
       -1.039498992,
       0.004560148530000001,
       -1.061859447e-006,
       1.144567878e-010,
       -4.763064160e-15}
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1922

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 24 Feb 2016 13:18 UTC**
That should definitely be looked at. Do you know for how many species this problem holds? Any more than the two mentioned in this ticket? Please note that the transformation from the database to the MSL was done about 15 or 16 years ago, and as far as I understand, data has also changed due to updated measurements (but not in this case). Today I would prefer to leave the database as is as a text file, and have a utility routine that reads in the data as needed, and one that generates the necessary Modelica-code.

modelica-trac-importer, 2017-01-15:
**Comment by veje&#x2e;johannes on 24 Feb 2016 14:22 UTC**
I don't know if it involves other species. However, the 100 to 200 other species I looked at in the database, column 33 to 48 was empty and should therefore not generate the same problem.

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 5 Apr 2016 08:23 UTC**

beutlich, 2017-02-01:
@hubertus65 First, I wanted to regenerate all records from the ThermoBuild database but could not find an up-to-date ASCII text file I could use for automatic processing. Thus, I counted the records with obvious transcription errors, that is where `blow[1]=0` and/or `bhigh[2]=0`, and was happy that only 18 gases are affected. I manually asked the ThermoBuild web interface (selecting H, D, C, O, Cl and Br from the table and then selecting the 18 gases from the list) to generate the desired thermodynamic data for these gases.
 
```
thermo                                                                          
   200.000  1000.000  6000.000 20000.000   9/09/04
CH2               D0(H2C-H): Ruscic,1999. Bunker,1983. Jacox,1998.              
 2 g 4/02 C   1.00H   2.00    0.00    0.00    0.00 0   14.0265800     390364.517
    200.000  1000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        10027.417
 3.218921730D+04-2.877601815D+02 4.203583820D+00 3.455405960D-03-6.746193340D-06
 7.654571640D-09-2.870328419D-12 0.000000000D+00 4.733624710D+04-2.143628603D+00
   1000.000  6000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        10027.417
 2.550418031D+06-7.971625390D+03 1.228924487D+01-1.699122922D-03 2.991728605D-07
-2.767007492D-11 1.051341740D-15 0.000000000D+00 9.642216890D+04-6.094739910D+01
CH3               D0(H3C-H): Ruscic,1999. Jacox,1998.                           
 2 g 4/02 C   1.00H   3.00    0.00    0.00    0.00 0   15.0345200     146658.040
    200.000  1000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        10366.340
-2.876188806D+04 5.093268660D+02 2.002143949D-01 1.363605829D-02-1.433989346D-05
 1.013556725D-08-3.027331936D-12 0.000000000D+00 1.408271825D+04 2.022772791D+01
   1000.000  6000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        10366.340
 2.760802663D+06-9.336531170D+03 1.487729606D+01-1.439429774D-03 2.444477951D-07
-2.224555778D-11 8.395065760D-16 0.000000000D+00 7.481809480D+04-7.919682400D+01
CH3OOH            Methyl Hydroperoxide, CH3-O-O-H. Dorofeeva,2001.              
 2 srd 01 C   1.00H   4.00O   2.00    0.00    0.00 0   48.0412600    -139000.000
    200.000  1000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        13918.692
-1.497974156D+05 2.656222273D+03-1.377060625D+01 6.588673830D-02-7.751801650D-05
 4.968800700D-08-1.314367640D-11 0.000000000D+00-3.058414201D+04 1.031696850D+02
   1000.000  6000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        13918.692
 3.060740610D+06-1.282959627D+04 2.541021168D+01-2.394481095D-03 4.443429910D-07
-4.416659500D-11 1.819673372D-15 0.000000000D+00 5.837492360D+04-1.387713096D+02
C2CL2             Hf:Manion,2002. Gurvich,1991 pt1 p114 pt2 p86.                
 2 g 5/02 C   2.00CL  2.00    0.00    0.00    0.00 0   94.9274000     226600.000
    200.000  1000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        14592.612
 3.871168730D+04-9.336870210D+02 1.119377832D+01-3.746287380D-03 6.770471350D-06
-4.910387450D-09 1.299068486D-12 0.000000000D+00 2.948153174D+04-3.310723530D+01
   1000.000  6000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        14592.612
 2.961995390D+05-2.044290845D+03 1.187529729D+01-5.115031340D-04 1.072666588D-07
-1.183417569D-11 5.326530480D-16 0.000000000D+00 3.636802500D+04-4.000826050D+01
C2CL4             Hf:Manion,2002. Gurvich,1991 pt1 p115 pt2 p88.                
 2 g 5/02 C   2.00CL  4.00    0.00    0.00    0.00 0  165.8334000     -24200.000
    200.000  1000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        19605.643
 3.746257500D+04-8.481774390D+02 1.009156041D+01 1.845463807D-02-2.390899731D-05
 1.514828471D-08-3.841119950D-12 0.000000000D+00-1.598289063D+03-2.368079134D+01
   1000.000  6000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        19605.643
-3.001289625D+05-1.132413909D+03 1.683073063D+01-3.288820170D-04 7.220975400D-08
-8.245417760D-12 3.810154710D-16 0.000000000D+00-2.292758158D+03-5.980164420D+01
C2CL6             Hf:Manion,2002. Gurvich,1991 pt1 p117 pt2 p90.                
 2 g 5/02 C   2.00CL  6.00    0.00    0.00    0.00 0  236.7394000    -148200.000
    200.000  1000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        27129.390
 1.593451902D+05-2.606697262D+03 2.028011918D+01 1.769364625D-02-3.133104894D-05
 2.389395021D-08-6.893037660D-12 0.000000000D+00-9.038007040D+03-7.953073010D+01
   1000.000  6000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        27129.390
-5.577351940D+05-4.976576790D+02 2.237042798D+01-1.477797229D-04 3.257641460D-08
-3.727218070D-12 1.723860648D-16 0.000000000D+00-2.335217699D+04-8.381736910D+01
C2HCL             Hf:Manion,2002. Gurvich,1991 pt1 p129 pt2 p99.                
 2 g 5/02 C   2.00H   1.00CL  1.00    0.00    0.00 0   60.4823400     226400.000
    200.000  1000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        11787.625
 1.329784931D+05-2.174542005D+03 1.497980787D+01-1.290343200D-02 1.602596399D-05
-9.152047430D-09 2.021222995D-12 0.000000000D+00 3.604801260D+04-5.958236750D+01
   1000.000  6000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        11787.625
 1.152149855D+06-4.461204350D+03 1.281367818D+01-6.797397460D-04 1.151395389D-07
-1.046657798D-11 3.949939170D-16 0.000000000D+00 5.233955000D+04-5.320661700D+01
C2HCL3            Hf:Manion,2002. Gurvich,1991 pt1 p134 pt2 p105.               
 2 g 5/02 C   2.00H   1.00CL  3.00    0.00    0.00 0  131.3883400     -17500.000
    200.000  1000.000 7 -2.0 -1.0  0.0  1.0  2.0  3.0  4.0  0.0        16604.624
[... cut ...]
 3.029467029D+05-1.079684654D+03 4.211417380D+00 6.260830070D-04-2.411665054D-07
 4.248425540D-11-2.387099102D-15 0.000000000D+00-1.188935343D+04-4.724898790D+00
END PRODUCTS                                                                    
END REACTANTS
```

Beside the transcription errors of `blow`, `ahigh` and `bhigh` for CH2, CH3, CH3OOH, C2CL2, C2CL4, C2CL6, C2HCL, C2HCL3, CH2CO_ketene, O_CH_2O, HO_CO_2OH, CH2BrminusCOOH, C2H3CL, CH2CLminusCOOH, HO2, HO2minus, OD, ODminus, also 'alow' was updated for C2CL2, C2CL4, C2CL6, C2HCL, C2HCL3, C2H3CL, CH2CLminusCOOH.

beutlich, 2017-02-02:
@hubertus65 Since the is fixed, I cam closing this one. Feel free to create a new enhancement issue for the general improvement on the management of the coefficients data.

closed 2017-02-02 by beutlich