modelica/ModelicaStandardLibrary #1927: Modelica.Icons.InternalPackage

modelica-trac-importer, 2017-01-15:
**Reported by stefanv on 26 Feb 2016 16:43 UTC**
There are a number of packages in the MSL and other libraries that extend from `Modelica.Icons.InternalPackage`.

When generating help for a library, should the contents of such a package be visible? The description invariably ends up being,

 This icon shall be used for a package that contains internal classes not to be directly utilized by a user.

because the extending package doesn't add its own documentation. However, there is no annotation suggesting that the package should not be visible. Is there a convention that says it shouldn't be (because it extends from `Modelica.Icons.InternalPackage`)? What do other tools do with such packages?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1927

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 8 Mar 2016 13:46 UTC**
Dymola doesn't do anything special for this for the help-pages. I think this is acceptable (tried to recreate it as good as possible - except the links), and think the text is useful.

\--
[[span(style=color: #007F00, Information )]]

Extends from Modelica.Icons.InternalPackage (Icon for an internal package (indicating that the package should not be directly utilized by user)).
[[span(style=color: #007F00, Package Content )]]

|**Name** |**Description**|
|-|-|
|Fet |Record for Fet parameters|
|FetModelLine|Record for Fet model line parameters|
|CurrrentsCapacitances|Currents and Capacities|
|fetRenameParametersDev|Device Parameter renaming to internal names|

\--

modelica-trac-importer, 2017-01-15:
**Modified by otter on 5 Aug 2016 05:58 UTC**

beutlich, 2017-08-21:
@svorkoetter Closing with worksforme.

beutlich, 2018-05-14:
@HansOlsson @svorkoetter We could set the `annotation(Protection(access=Access.icon))` for these Internal packages to explicitely exclude them from HTML generation. What do you think?

HansOlsson, 2018-05-14:
To me it is still not clear that there is a problem with having documentation for the classes in the internal package.

As an example consider 
Modelica.Blocks.Continuous.Internal.Filter.Utilities.BesselBaseCoefficients used in
Modelica.Blocks.Continuous.Internal.Filter.base.Bessel used in
Modelica.Blocks.Continuous.Filter
to me it makes sense that one can dig down to BesselBaseCoefficients to understand the underlying calculations; without having all that documentation in the Filter-model.

And I don't see a good way to document the models in Modelica.Blocks.Continuous.Internal without having some documentation for that package.

beutlich, 2018-05-14:
> As an example

Enough reason to include the generated HTML documentation of Internal classes as is.

closed 2018-05-14 by beutlich