modelica/ModelicaStandardLibrary #1941: ModelicaStandardTables_* fails for restarted simulations

modelica-trac-importer, 2017-01-15:
**Reported by otter on 3 Mar 2016 04:41 UTC**
If simulation is halted and restarted afterwards, then this fails if models of Modelica.Blocks.Sources.CombiTimeTable or Modelica.Blocks.Tables are present in the model. The reason is that tables are read during initialization and restarting a model does not execute the initialization section.

This is a branch of #1899, to temporarily fix this issue on a branch.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1941

modelica-trac-importer, 2017-01-15:
**Comment by otter on 3 Mar 2016 04:52 UTC**
Fixed C-code in r9121 on branch https://svn.modelica.org/projects/Modelica/branches/development/ModelicaStandardTables-Fix1899 by the suggestion from Hans in #1899.

Object libraries need still to be generated

modelica-trac-importer, 2017-01-15:
**Comment by otter on 3 Mar 2016 07:12 UTC**
Further fix in r9124, by including optional table-read calls in all get-functions

modelica-trac-importer, 2017-01-15:
**Comment by otter on 3 Mar 2016 10:43 UTC**
Object libraries added in r9127

modelica-trac-importer, 2017-01-15:
**Comment by sjoelund&#46;se on 3 Mar 2016 16:37 UTC**
I added #1950 to propose adding a save/restore interface instead of needing changes to C-sources for initialization of tables

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 23 Jun 2016 21:37 UTC**

beutlich, 2017-04-10:
@MartinOtter I suppose we can savely close this split issue.

closed 2017-04-10 by beutlich