modelica/ModelicaStandardLibrary #1942: Resource leak in Modelica.Utilities.Examples.readRealParameterModel (long term edition)

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 3 Mar 2016 07:52 UTC**
There are Modelica tools that evaluate parameter/variable expressions in the GUI **before** simulation. In case of _Modelica.Utilities.Examples.readRealParameterModel_ this means that parameter _J_, _phi_rel0_ and _w_rel0_ are determined by a function calls of _readRealParameter_ and thus by reading the values from file. It turns out that the file Examples_readRealParameters.txt is opened but not closed again.

See also #1940.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1942

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 14 Mar 2016 13:57 UTC**

AHaumer, 2019-07-19:
This is still an issue. Well in the meantime I have my own implementation but I'd like to see a clean implementation by womeone who's really capable of C-programming. 

1. The file should be closed after reading.
2. It should be a user's choice whether the parameters are read during translation or during execution.

MartinOtter, 2019-07-22:
> 
> 
> This is still an issue. Well in the meantime I have my own implementation but I'd like to see a clean implementation by womeone who's really capable of C-programming.

Why C? The example implementation is in Modelica (not in C).

>     1. The file should be closed after reading.

Yes. Just add Streams.close(fileName) at the end of the function.

> 
>     2. It should be a user's choice whether the parameters are read during translation or during execution.

By purpose, function readRealParameter(..) was placed under Examples to demonstrate the concept. Sure, whenever something is not possible in current Modelica, one can implement this in C. But then there are a lot of issues with regards to support of different operating systems and different tool vendors. A pure Modelica implementation would be much more robust against these practical details and could be much easier adapted by a user to his/her needs.

Therefore, an alternative is to add a Modelica annotation, that performs the reading at run time and not during compile time (and just keep the current Modelica code).

AHaumer, 2019-07-23:
@MartinOtter 
I have to admit that I didn't look into that functions for some time ...
In the past, the bug was buried in readLine:
 external "C" string = ModelicaInternal_readLine(fileName,lineNumber,endOfFile) annotation(Library="ModelicaExternalC");
To be concise, I have my own clumsy C-implementation of readLine (without "cache").

Maybe the "close" is fixed now. The question is whether the file is read during translation or during execution. 

I'd have to test whether it is possible to perform a simulation, change the file and re-run the simulation without re-translate. This is needed e.g. to perform optimization with an external optimzer like GenOpt.