modelica/ModelicaStandardLibrary #1954: Support directories in loadResource(..)

modelica-trac-importer, 2017-01-15:
**Reported by otter on 9 Mar 2016 18:40 UTC**
At the 89th Modelica Design Meeting the following proposal was worked out:

 It is proposed to generalize the available ModelicaService function “loadResource(..)”. If the URI or local path name identifies a directory (= collection of resources), then the absolute path name of this directory is returned. The returned absolute path of a directory does not have a trailing slash [_If the URI would specify data in another form (say a database), then all resource elements identified by the URI could be copied to a temporary, local directory and the absolute path name of this directory is returned_]. 

This feature is already supported by Dymola, JModelica.org, OpenModelica and by SystemModeller.

It has to be checked whether the "not having a traling slash" is fine. Otherwise, there seems to be agreement to introduce this feature in the next version of MSL (Modelica.Utilities.Files.loadResource()) and correspondingly in ModelicaServices.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1954

modelica-trac-importer, 2017-01-15:
**Comment by otter on 9 Mar 2016 22:50 UTC**
In Dymola 2017 Beta.1: loadResource(..) only prepends the heading path to the URI, but does not append something. As a result
```
loadResult("modelica://Modelica/Resources")  -> "...trunk/Modelica/Resources"
loadResult("modelica://Modelica/Resources/") -> "...trunk/Modelica/Resources/"
```
Seems to be a good semantics

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 15 Apr 2016 11:17 UTC**
From #1976:
> Usually, directory names are distinguished from file names by a trailing slash.

Thus I would prefer to explicitly have a trailing slash for directory names.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 18 Apr 2016 07:47 UTC**
Replying to [comment:2 beutlich]:
> From #1976:
> > Usually, directory names are distinguished from file names by a trailing slash.
> Thus I would prefer to explicitly have a trailing slash for directory names.

Since the last part of the directory is provided by the user, an option seems to be to not change this part and only adapt the leading part, so
```
parameter String dir1 = "modelica://Modelica/Resources/Data";
parameter String dir2 = "modelica://Modelica/Resources/Data/";
final parameter String fullDir1 = loadResource(dir1);
final parameter String fullDir2 = loadResource(dir2);

/* On windows the following is returned ("/" is treated as "\"):
   fullDir1 = "C:/svn.modelica.org/Modelica/Resources/Data";
   fullDir2 = "C:/svn.modelica.org/Modelica/Resources/Data/";
*/
```

The tool vendors that support directories for loadResource should give their opinion.

modelica-trac-importer, 2017-01-15:
**Modified by otter on 5 Aug 2016 05:59 UTC**

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 13 Sep 2016 11:47 UTC**
An issue is that 
"modelica://Modelica.Mechanics.Rotational" is a reference to the class
"modelica://Modelica.Mechanics.Rotational/Resources/A.jpg" references a jpg
"modelica://Modelica.Mechanics.Rotational/Resources/" references a directory
"modelica://Modelica.Mechanics.Rotational/Resources" references a directory
"modelica://Modelica.Mechanics.Rotational/" is this a directory?

Should trailing slashes be preserved for directories as arguments to loadResource?
Possibilities: Forbid trailing slash, preserve trailing slash, (remove it if present).
Forbid trailing slash: 1
Preserve trailing slash: 4
Remove it if present: 1
Always add trailing slash: 2

One possibility is to require that the path-name should be non-empty, otherwise it is an error. Can then use "modelica://Modelica.Mechanics.Rotational/." or "modelica://Modelica.Mechanics.Rotational/./" to reference the directory.
Require non-empty path: 2
Allow empty path - returning the directory: 3 (with slash)

Henrik: Check common libraries for files - and what they return for normalization. (Boost, Qt, ...):
Windows API _splitpath return trailing slash for directory, but other routines are without trailing slash.
Qt's QDir function don't add trailing slash
Boost does not seem to add trailing slash
Note that on Windows the current directory function returns "c:\" and "c:\Dir" - because "c:" wouldn't always work. However, loadResource will not return a drive (unless ".." is used as path - but then ".." should be in the result).

Would be good if agreement. 
Note: 
IncludeDirectory and LibraryDirectory are defined without trailing slash.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 13 Sep 2016 12:11 UTC**
Conclusion - remove trailing slash (if any) - since common on Unix and not always present on Windows (note that this can be done without checking if it is directory - just always remove the trailing slash).
Favor: 6
Against: 0

And similarly without slash for "modelica://Modelica.Mechanics.Rotational/".

"modelica://Modelica.Mechanics.Rotational/" -> ".../dir"
"modelica://Modelica.Mechanics.Rotational/B.jpg" -> ".../dir/B.jpg"
"modelica://Modelica.Mechanics.Rotational/A/" -> ".../dir/A"
"modelica://Modelica.Mechanics.Rotational/A/B.jpg" -> ".../dir/A/B.jpg"

beutlich, 2018-03-21:
What is left to do? Documentation only?

beutlich, 2018-08-07:
rescheduling milestone

beutlich, 2019-10-28:
removing milestone